Open-Smartwatch/open-smartwatch-os #2: Compiling for Windows

liyanboy74, 2021-03-02:
I use VS Code in windows10 & installed PlatformIO,
Do I have to make certain settings before compiling the code? 
for Example add some Define?
or add another lib?

**Errors:**
```
.pio\libdeps\pico32_LIGHT_EDITION\TFT_eSPI\Processors/TFT_eSPI_ESP32.h:79:36: fatal error: SPIFFS.h: No such file or directory
.
.
.
src\apps\_experiments\runtime_test.cpp:10:23: error: 'MINI_IOT_DEVICENAME' was not declared in this scope
src\apps\_experiments\runtime_test.cpp:10:44: error: 'MINI_IOT_SERVER' was not declared in this scope
src\apps\_experiments\runtime_test.cpp:10:61: error: 'WIFI_SSID' was not declared in this scope
src\apps\_experiments\runtime_test.cpp:10:72: error: 'WIFI_PASS' was not declared in this scope
```

uvwxy, 2021-03-02:
Indeed, please copy the file `include/config.h.example` to  `include/config.h` and adapt the values according to your requirements. I will add this to the readme today.

liyanboy74, 2021-03-02:
That's right, thank you.

commit 8bcfdda5 (2021-03-03): Merge pull request #3 from liyanboy74/master

close #2; update readme
commit d5b7c320 (2021-03-03): close #2; update readme.md
closed 2021-03-03 by OneGneissGuy