Open-Smartwatch/open-smartwatch-os #4: adcStart not declared

liyanboy74, 2021-03-09:
Error:
`src/hal/power.cpp:59:17: error: 'adcStart' was not declared in this scope`

I think `adcStart()` Removed from new update [ESP32 Arduino Release 1.0.5 based on ESP-IDF v3.3](https://github.com/espressif/arduino-esp32/releases/tag/1.0.5)

[As shown here!](https://github.com/espressif/arduino-esp32/pull/3377/files/51184d7d8a48c9b0dd5ba0e9545f52aa51271cbf#diff-69413c45ba9a7c93f4bb2441433dd75b00075174cb15e5606b3335eeef3d8d24L117)

But I'm not sure what to use instead !

uvwxy, 2021-03-09:
Just for comparison what I have done so far:

Install Visual Studio Code
Install the PlatformIO Extensions

Open the source folder with visual studio code.

Are we doing the same things here? Maybe my visual studio code extension is using an older version os ESP-IDF.

uvwxy, 2021-03-09:
This is my build output

```
> Executing task: platformio run <

Processing pico32_LIGHT_EDITION (platform: espressif32; board: pico32; framework: arduino)
-------------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/pico32.html
PLATFORM: Espressif 32 (2.0.0) > ESP32 Pico Kit
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (esp-prog) External (esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 3.10004.200129 (1.0.4) 
 - tool-esptoolpy 1.20600.0 (2.6.0) 
 - toolchain-xtensa32 2.50200.80 (5.2.0)
```

Could you try again with ESP-IDF 3.10004.200129 (1.0.4) ?

uvwxy, 2021-03-09:
I will update the code to work with ESP-IDF 3.3 / 1.0.5 though, in the next days.

liyanboy74, 2021-03-09:
in file `.platformio\packages\framework-arduinoespressif32\cores\esp32\esp32-hal-adc.h` not found function `adcStart()` ,
but it's in `.platformio\packages\framework-arduinoespressif32@3.10004.210126\cores\esp32\esp32-hal-adc.h` file.
Definitely due to the update !

neofuturism, 2021-03-09:
@liyanboy74 I have the same issue, I'm an absolute beginner with Vscode, could you tell me how you reset your ESP-IDF version, Thanks

uvwxy, 2021-03-09:
@neofuturism @liyanboy74 

I removed the getBatteryVoltage function (for now), please check if the latest commit works with the new platform. changing the ESP-IDF version is not the way forward.

liyanboy74, 2021-03-09:
SUCCESS Build.

neofuturism, 2021-03-09:
thanks @uvwxy, build and upload was a success!
but now the display is dark and it reboots, 
Here is the output of from the serial monitor.

> Rebooting...
> ets Jun  8 2016 00:22:57
> 
> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
> configsip: 188777542, SPIWP:0xee
> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
> mode:DIO, clock div:2
> load:0x3fff0018,len:4
> load:0x3fff001c,len:1044
> load:0x40078000,len:10124
> load:0x40080400,len:5828
> entry 0x400806a8
> E (28) RTC_MODULE: /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rtc_module.c:163 (rtc_gpio_deinit):RTC_GPIO number error
> [E][esp32-hal-cpu.c:115] removeApbChangeCallback(): not found func=400D9B18 arg=3FFBDBDC
> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
> Core 1 register dump:
> PC      : 0x4014b910  PS      : 0x00060d30  A0      : 0x800d4ccd  A1      : 0x3ffcbaa0  
> A2      : 0x3ffba4f4  A3      : 0x00000008  A4      : 0x08000000  A5      : 0x3ffc1e68  
> A6      : 0xffffffff  A7      : 0x00000001  A8      : 0x8014b6de  A9      : 0x3ffcba80  
> A10     : 0x3ffba4f4  A11     : 0x000001ff  A12     : 0x3ff65000  A13     : 0x00000031  
> A14     : 0x00060d20  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
> EXCVADDR: 0x00000008  LBEG    : 0x400d71b5  LEND    : 0x400d71c2  LCOUNT  : 0x00000000  
> 
> ELF file SHA256: 0000000000000000
> 
> Backtrace: 0x4014b910:0x3ffcbaa0 0x400d4cca:0x3ffcbac0 0x400d4ede:0x3ffcbae0 0x400d431b:0x3ffcbb00 0x400d4328:0x3ffcbb20 0x400d2c49:0x3ffcbb40 0x400d2ff1:0x3ffcbb60 0x400d381f:0x3ffcbb80 0x400da73a:0x3ffcbbb0 0x4008a582:0x3ffcbbd0

neofuturism, 2021-03-09:
I have selected it through PIO configuration, but is there another way to select the right target?
Sadly, the exception decoder return nothing :/ 

[image]
[image]
[image]

neofuturism, 2021-03-09:
ok cool, here's the output
PC: 0x4014b910
EXCVADDR: 0x00000008

Decoding stack results
[image]

OneGneissGuy, 2021-03-09:
Can we lock the examples to a version ?

neofuturism, 2021-03-17:
Hello @uvwxy, I deleted the folder and reuploaded it,
everything works perfectly now :)

closed 2021-04-06 by uvwxy