Open-Smartwatch/open-smartwatch-os #13: App: Tide Estimation

uvwxy, 2021-04-07:
Ocean tide prediction is actually not trivial and involves the respective data set (tide coefficients) for a single location.

I have written a blog post about this a while ago:

https://p3dt.net/post/2019/03/10/predicting-tides.html

Tomaskocarter, 2021-06-19:
This is a good idea, and for us land-locked users, we could do something similar but with sunrise and sunset times.

uvwxy, 2021-09-06:
Here are my ideas:

I think I'm aiming for a widget as a first step. A 24h (0h00-23h59) histogram of water level, and a red line indicating the current time. no number, no values, just a quick glance to see where where we are at during the day regarding tides.

Next step would be a dedicated tide app.
- maybe: a location selector (we should do some more work on reusable UI widgets first, like a selection list), this could be done automatically on the GPS edition some time later. also not all tide coefficients fit into the flash memory, so a single selection at compile time  will be the starting point
- a detailed tides screen, that:
  - shows the location name (each set of tide coefficients corresponds to a specific location/buoy)
  - 24h histogram with with water levels, lines marking metres (m)
  - short click on btn 1 shows daily tides
  - btn 2 + btn 3 cycle through days (we can calculate the tides for any days with the harmonics)
  
 Algorithmic Todos:
 Implement tide estimation algorithm in c/c++, as documented: https://p3dt.net/post/2019/03/10/predicting-tides.html
 Implement a converter in js/bash to fetch tide harmonics and translate into a C struct(?). This struct is the input to the tide estimation. (technically an array of triples)

Later for GPS edition:
download and/or parse harmonics form text file:
example: https://webcritech.jrc.ec.europa.eu/worldsealevelinterface/Default.aspx?id=1284
harmonics data should reside on the micro sd card

closed 2022-02-05 by uvwxy