Open-Smartwatch/open-smartwatch-os #16: Get LVGL to render into Graphics2D

uvwxy, 2021-04-07:
As a developer I want to be able to develop UIs using LVGL on desktop to create example applications as well as for arduino to run on the smartwatch. developing on desktop first greatly improves the development time.

Proof of concept:

https://www.youtube.com/watch?v=_QIXN23bQsM

uvwxy, 2021-04-07:
What's the status @VedantParanjape ?

VedantParanjape, 2021-04-07:
@uvwxy I didn't work on this as smartwatch, but I am very comfortable with lvgl now. Here's my code: https://github.com/Open-Authenticator/open-authenticator-app

I will try to make a POC soon.

uvwxy, 2021-04-08:
Looking forward to the discussion on discord later. Would be great if you could provide the direction how to adapt to use the lvlg code in a practical way!

VedantParanjape, 2021-04-09:
> Looking forward to the discussion on discord later. Would be great if you could provide the direction how to adapt to use the lvlg code in a practical way!

Yes, lvgl has a lot of boiler plate. But thing is it runs on a single thread, and you need to use their lvgl_task_run to run tasks handling widgets. You can't actually call functions which take time inside lvgl loop, or else the UI freezes. What I did was write a event handler in esp-idf, check here: https://github.com/Open-Authenticator/open-authenticator-app/blob/main/main/gui_event_handler.c

So, now I only post events from lvgl thread, and receive notifications back when it is completed.

uvwxy, 2021-04-09:
how/where do you handle full screen buffering?

VedantParanjape, 2021-04-09:
From here, this function

https://github.com/Open-Authenticator/open-authenticator-app/blob/922acd76e1100d7369b28fa62c5de50bdcee0947/main/lvgl_gui.c#L325

uvwxy, 2021-05-16:
No one seems to be actively working on this.

closed 2021-05-16 by uvwxy