Open-Smartwatch/open-smartwatch-os #17: Extend Graphics2D to Circular2DGraphics2D to improve memory consumption

uvwxy, 2021-04-07:
Currently we waste around 25% of buffer for the invisible corners around the 240*240 circular display.

Two places that need to be changed for this to work is:

- the constructor needs to change the width of the allocated chunks
- drawPixelPreclipped needs to be adapted to dynamic chunk sizes

uvwxy, 2021-04-07:
I need help: it works, but as soon as the memory is freed it crashes.

My assumption is that the chunk size calculation is not correct and it writes across the chunk arrays boundary

VedantParanjape, 2021-04-07:
@uvwxy can you add relevant LOC here ?

uvwxy, 2021-04-08:
https://github.com/Open-Smartwatch/lib-open-smartwatch/blob/master/gfx_2d.h

Lines: 
* in `enableBuffer()` -> the `if(isRound)` branch
* in `drawPixelClipped()`-> the `if (isRound && isInsideChunk(x, y))` branch
* `disableBuffer` is where the exception flies

I'm not sure where exactly I'm going wrong. Probably have been staring at it for too long and don't see it anymore.

VedantParanjape, 2021-04-09:
> buffer = new uint16_t*[numChunks];

What does this do? I read the code, and what you are doing is kind of an array of pointers which point to an array of pixel rows ?

Can you do something like allocate a single big buffer and then use arithmetic to find pixel positions in the linear array ?

uvwxy, 2021-04-09:
The chunking is part of the unoptimized version because the esp32 does not provide the option to allocate arrays of an arbitrary size (even if enough memory is free).

this optimisation is only an approximation, and not optimal, but still achieves to use 20% less memory, while still using chunks to be able to allocate everything

VedantParanjape, 2021-04-09:
> The chunking is part of the unoptimized version because the esp32 does not provide the option to allocate arrays of an arbitrary size (even if enough memory is free).
>
> this optimisation is only an [...]

I can do this on esp-idf, idk why it doesn't work with arduino

commit 3fcb8659 (2021-05-16): close #17: use round framebuffer
closed 2021-05-16 by uvwxy