Open-Smartwatch/open-smartwatch-os #18: ClearFont option in gfx_2d_print

max9901, 2021-04-07:
add a way to clear a custom font. 

//to reset the font. 
Graphics2DPrint::clearfont();

This is my first "larger" github project and i will use this fairly simple issue as a way to see if i get the gist of the workflow.

uvwxy, 2021-04-07:
I re-opened this ticket. I will close it once it has been merged into the source code. Please create a pull request against lib-opensmartwatch

max9901, 2021-04-07:
I decided to leave it out because u can just use

Graphics2DPrint::setFont(NULL).
 
Which is not as user-friendly for a beginning Arduino user but I thought that's probably not our target audience anyway with the lib?

max9901, 2021-04-07:
Which btw doesn't guarantee that the cursor is put in the right place I think, but we can probably merge things like this into a rewrite of the Graphics2DPrint class anyway?

uvwxy, 2021-04-08:
Ok, we can add this with #15

closed 2021-04-08 by uvwxy