Open-Smartwatch/open-smartwatch-os #19: Initialize the framebuffer ?

max9901, 2021-04-09:
Shalle we change malloc to calloc ?

or initialize the buffer to 0 imediatly when creating the buffer.

uvwxy, 2021-04-11:
I will try this out.

max9901, 2021-05-15:
it is fixed when u change:

`Graphics2D::enableBuffer()`

 `buffer[i] = (uint16_t*)malloc(width * chunkHeight * sizeof(uint16_t));`
 into 
` buffer[i] = new uint16_t[width * chunkHeight]();`
 
```
 new int[x]   -> not initialized 
 new int[x]() -> initialized to 0
 
```

uvwxy, 2021-05-16:
Does this have any performance drawbacks?
Is it really required, especially as all/most apps do a `hal->gfx()->fill(...)` at the beginning?

max9901, 2021-05-16:
No but i was hoping to get an initial face startup to go a little faster. By allowing for a basic first render at the setup of a face. 
So for example a digital watchface only loads the time at setup. and all other features will get added in the loop().
but tbh didn't really note a difference when experimenting yesterday with it.

'new[]()' is ofcourse a bit slower but it is faster then the 'gfx()->fill()' call and this way we are sure we never read un-initialed memory. Which adds value in my opinion.

uvwxy, 2021-05-16:
Ok, I will change it accordingly

commit 66ad8af2 (2021-05-16): fix #19: initialise 2d gfx framebuffer
closed 2021-05-16 by uvwxy