Open-Smartwatch/open-smartwatch-os #23: Fitness tracking

crazycaleb2008, 2021-04-10:
There should be a screen that shows steps, calories burned,goals and other things fitness related. 🤷‍♂️

uvwxy, 2021-04-10:
Does anyone know some good average values for kCal / step ?

crazycaleb2008, 2021-04-10:
We were talking about it on the discord

matti04, 2021-04-10:
Some results of our previous discussions: 
-Calculate speeds with steps/second
-Enter step length for 3 different speeds in the configuration file
-Use an averaged value
-https://www.insider.com/how-many-steps-a-day-to-lose-weight
-GPS edition might even take the height difference into account, assuming that is possible
-Could be difficult to run it in deep-sleep mode

crazycaleb2008, 2021-04-10:
[image]

Heres a concept i made. Had to do it on my phone so dont judge me 😂

cawolfkreo, 2021-04-11:
I saw the comments about this screen on the discord and tried to help with the design of @crazycaleb2008 and I came up with these designs: 
[image]
The idea was to distance ourselves a bit from the apple watch by using semicircles instead of a full circle. The data this design needs is:
* **Steps counter:** Just the amount of steps taken.
* **Distance walked:** how far (aprox) has the user walked/run, either using `steps * average step distance` and/or the GPS (as you guys mentioned previously).
* **Time since the fitness tracking started:** you count in seconds and display it in minutes.
* **kCal burned:** calculated as @matti04 pointed out and the unit will be `Cal` or `Calories` instead of `kCal` since people are used to assuming kCal = Cal in most food products, packaging, etc. 

### You can see the full designs, comments for each of them and a small "how this will be used" [here](https://www.figma.com/file/hf30vruZyHyqMnJ8l1sqPH/Open-Smart-Watch-fitness-face?node-id=43%3A2)

uvwxy, 2021-04-11:
Currently from those only works:
- steps counter

Regarding Distance walked:
- the BMA400 supports activity recognition: still, walking, running
- depending on activity mode walking and running one could use two average step lengths to estimate the distance run. allthough this will be error prone, it might give an interesting estimation.
- Average walking step length could be estimated with "For women, simply multiply 0.413 by your height in centimeters. Men should multiply 0.415 by their height in centimeters." (See for example https://livehealthy.chron.com/determine-stride-pedometer-height-weight-4518.html)

Regarding kCal burned:
- "A typical 160-pound person of average height will burn about 40 calories per 1,000 steps. This is the equivalent of 0.04 calories per step." (See for example https://www.verywellfit.com/pedometer-steps-to-calories-converter-3882595)
- find something similar for running.

Although those values are just estimation, they might be a good starting point.
The steps are being tracked by the BMA400 even while the watch is in deep sleep, this means we can create statistics with out any power drawbacks over a day.

For exercising I would recommend to implement the watch to respond to activity change interrupts, wake up on change and keep adding the step swalked or steps run.

Tomaskocarter, 2021-06-19:
Personally I really like the watch face that's second from the right on the bottom row. I think making the time larger so that it is easy to read at a quick glance, and adding in the date could be an improvement, but overall I think the simplicity of it gives it a minimalistic look that's pleasing to the eye.

RuffaloLavoisier, 2022-04-03:
Is this still going on? How is the current situation?

RuffaloLavoisier, 2022-05-20:
See https://github.com/Open-Smartwatch/open-smartwatch-os/pull/223

simonmicro, 2022-05-31:
What do you guys @crazycaleb2008 and @matti04 think? With the current `develop` branch @RuffaloLavoisier implemented everything beside goals. Should we close this issue or still keep it open?

closed 2022-05-31 by simonmicro

crazycaleb2008, 2022-05-31:
This looks great! Will have to bust out my osw to try it. It dosent matter to me if you keep it open or close 😜

RuffaloLavoisier, 2022-05-31:
@crazycaleb2008 Thank you for the idea. And I was able to create it.