Open-Smartwatch/open-smartwatch-os #27: Feature requests and contrib

realrk69, 2021-04-13:
So eventually, this app will need support of a companion app to sync data to the cloud or to a smartphone.

I have a few suggestions in this regard:

1. Creating a standard format in which the data can be converted and loaded  or streamed into apps to keep a historical track.

2. Companion app should have features such as ability to provide OTA updates directly to the user. This will require some hardware mods such as the inclusion or a I2C or SPI parent controller which will write the new streamed code.

3. I recommend creating this app using a cross platform technology like a progressive web app or flutter. I can create a few pull requests if this interests you. But please be specific about what features will be expected from the app (tracking exercises, acitivity, sleep etc.)

4. Heart rate tracking: with my experience in optics and mechanical design, and the possibility of having an added SPI connection, I can suggest a few accurate but affordable heart rate sensor modules.

SeanReg, 2021-04-13:
At the moment I made this Basic Android App to test the Bluetooth LE Service. App connects to device and is able to send Android Notifications to the watch. Could be reimplemented in cross-platform framework if needed

https://github.com/SeanReg/open-smartwatch-android-companion

rasppliances, 2021-04-28:
> At the moment I made this Basic Android App to test the Bluetooth LE Service. App connects to device and is able to send Android Notifications to the watch. Could be reimplemented in cross-platform framework if needed
> 
> https://github.com/SeanReg/open-smartwatch-android-companion

Is there an APK for this?  What has to change in the OSW software for this to work?

uvwxy, 2021-04-29:
@rasppliances This is still in a proof of concept phase. The short answer is: a lot has to be done to fully support this. Also: keeping bluetooth on drastically reduces the available memory on the watch, and the screen can not be fully buffered, reducing the frame rate. It's nice to have a starting point provided by @SeanReg , but I assume it's still very early in this project to see this being implemented as one would expect it to work.

I'm not going to look deeper into this, at the moment. Maybe this will be more relevant once the GPS edition hardware is ready.

@realrk95 could you split your requests into several issues, such that each topic can be tracked separately? (This is why I close the ticket)

closed 2021-04-29 by uvwxy