Open-Smartwatch/open-smartwatch-os #35: Possibility to develop without the board?

ffenix113, 2021-04-19:
Hello, 

This seems like an awesome and interesting project, and I personally would be interested to play around and maybe even contribute.

But it would be nice to know if development and testing(at least somethung) can be achieved without having a board? For example with Qemu or similar.

euanlc, 2021-04-19:
Hi, if you look at paul's instagram he has a post where he appears to be using an emulator. If you want information about that from him your best bet is to join the [discord](https://discord.gg/9QQuv6ta) to talk to him directly.
hope this helps,
euan

cawolfkreo, 2021-04-19:
There is an amulator of sorts you can use as @euanlc mentioned, you can find it here: https://github.com/Open-Smartwatch/lib-open-smartwatch

zbot473, 2021-04-19:
Someone would have to write an entire emulator with bindings to methods that fake or use the host's Wi-Fi/Bluetooth/Accelerometers/GPS/SD/Buttons

ffenix113, 2021-04-24:
Thank you for the info. 

I will check out the graphics repo.

closed 2021-04-24 by ffenix113