Open-Smartwatch/open-smartwatch-os #43: rtc_module.c:155 (rtc_gpio_deinit):RTC_GPIO number error

judomat, 2021-04-25:
#### When uploading the latest version of open-smartwatch-os I get this exception: 
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5828
entry 0x400806ac
E (36) RTC_MODULE: /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/rtc_module.c:155 (rtc_gpio_deinit):RTC_GPIO number error
```
And when I try to connect to the WiFi my serial monitor crashes with this exception:
```
Exception in thread rx:
Traceback (most recent call last):
  File "/Users/matteo/.platformio/penv/lib/python3.9/site-packages/serial/serialposix.py", line 575, in read

--- exit ---
    buf = os.read(self.fd, size - len(read))
OSError: [Errno 6] Device not configured

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Users/matteo/.platformio/python3/lib/python3.9/threading.py", line 954, in _bootstrap_inner
    self.run()
  File "/Users/matteo/.platformio/python3/lib/python3.9/threading.py", line 892, in run
    self._target(*self._args, **self._kwargs)
  File "/Users/matteo/.platformio/penv/lib/python3.9/site-packages/serial/tools/miniterm.py", line 499, in reader
    data = self.serial.read(self.serial.in_waiting or 1)
  File "/Users/matteo/.platformio/penv/lib/python3.9/site-packages/serial/serialposix.py", line 581, in read
    raise SerialException('read failed: {}'.format(e))
serial.serialutil.SerialException: read failed: [Errno 6] Device not configured
```
And the watch restarts.

I saw that there is a closed issue similar to this one: #4 
That issue was solved by changing the platform version from 3.1 to 3.0.
I have tried platform version 3.2, 3.1 and 3.0 but I always get the same results.

uvwxy, 2021-04-27:
Hi @judomat 

I can fix the first part. Regarding the second, I think this is an issue with your notebook/USB port:

https://esp32.com/viewtopic.php?t=2595

googling `raise SerialException('read failed: {}'.format(e)) serial.serialutil.SerialException: read failed: [Errno 6] Device not configured` will show you more threads/posts/... going deeper into that issue.

uvwxy, 2021-04-27:
feel free to discuss this on discord https://discord.gg/9DK5JY6 with us :)

commit 6ab87ec3 (2021-04-27): fix #43: don't deinit gpios
closed 2021-04-27 by uvwxy