Open-Smartwatch/open-smartwatch-os #49: Twist wrist to turn on display

rasppliances, 2021-04-28:
Hello,

I was looking at the accelerometer to see if I could add a twist wrist to turn on display feature like you see in the "Major Brands", but I'm not sure it can be done the way the OSW OS is written.

I noticed that in main.cpp, if the app is the watchface app, and the timeout interval hits, it puts the system into deep sleep with

hal->deepsleep()

And that in turn eventually calls the line

esp_deep_sleep_start()

which puts the esp unit into a deep sleep that can only be woken up on GPIO_NUM_0 going low.

If I understand all of this right, which would surprise me if I did, that means that the system isn't doing anything except waiting for that GPIO_NUM_0 to go low, and all the shaking and twisting in the world won't make a difference.

But then I think about the step counter.  That's got to be running while the display is off, otherwise it's no good.

Is it possible to wake the unit from something other than GPIO_NUM_0 going low when the unit is in deep sleep?

uvwxy, 2021-04-29:
This can be done with proper configuration of the BMA400 sensor and responding to the interrupts. This way the CPU can remain in deep sleep to conserve power. 

The same way the step counting is done by the BMA400 itself. the number of steps is read by the CPU at runtime form the BMA400 via I2C.

rasppliances, 2021-04-29:
Interesting.  I'm going to look into that.  Not that I'll be able to do anything about it, but I'm going to look.  Thanks for the reply. Should we close this issue?

rasppliances, 2021-05-06:
I've actually got this working.

In sensors.cpp, at line 175 I added

` rslt = bma400_get_sensor_conf(accel_setting, 3, &bma);
  bma400_check_rslt("bma400_get_sensor_conf", rslt);

  // get current state of 0x1F register
  uint8_t regSet = 0;
  rslt = bma400_get_regs(0x1f, &regSet, 1, &bma);
  bma400_check_rslt("bma400_get_regs 0x1f", rslt);

  if (rslt == BMA400_OK) {
    // we were successful getting the 0x1f register

    // set orientation change used for interrupt
    uint8_t data=BMA400_AXIS_X_EN & BMA400_DATA_SRC_ACCEL_FILT_LP;
    rslt = bma400_set_regs(0x35, &data, 1, &bma);

    // set the threshold for the twist
    data=0x7F;
    rslt = bma400_set_regs(0x36, &data, 1, &bma);
    bma400_check_rslt("bma400_set_regs 0x36", rslt);

    // set stable time in 50ths of a second

    data=0x19;
    rslt = bma400_set_regs(0x38, &data, 1, &bma);
    bma400_check_rslt("bma400_set_regs 0x38", rslt);

    // add orientation change to current interrupt settings
    regSet = regSet & BMA400_AXIS_X_EN;
    rslt = bma400_set_regs(0x1f, &regSet, 1, &bma);
    bma400_check_rslt("bms400_set_regs 0x1f", rslt);

    // get the current setting for 0x21
    rslt = bma400_get_regs(0x21, &regSet, 1, &bma);
    bma400_check_rslt("bma400_get_regs 0x21", rslt);

    // add orientch to int1 map

    if (rslt == BMA400_OK) {
      regSet = regSet & 0x02;
      rslt = bma400_set_regs(0x21, &regSet, 1, &bma);
    } else {
      data=0x02;
      rslt = bma400_set_regs(0x21, &data, 1, &bma);
    }
  }
`

Then, in power.cpp at line 107, I added

`esp_sleep_enable_ext0_wakeup(GPIO_NUM_34, HIGH);`

xasz, 2021-05-12:
If you have it working you can do a pull request :)

rasppliances, 2021-05-12:
> If you have it working you can do a pull request :)

I've never done that before, so I'm not sure I did it right, but I think I made a pull request with my changes.

commit 900f9b69 (2021-05-16): fix #49, add flag to draw high quality thickline/arcs, add brightness config key
closed 2021-05-22 by uvwxy