Open-Smartwatch/open-smartwatch-os #54: Implement proper storage HAL

Simonmicro, 2021-05-01:
This should store some kind of class instance (so just write it directly to the memory; read it directly on boot).
Possible store some things like:
* Last open app (so when we crash again, the user will not notice too much)
* Last connected BLE device (to reconnect on boot)
* App data? Stopwatch stuff?

But we should somehow minimize the write cycles on this part (only static data?), otherwise we literally kill the memory over time...

uvwxy, 2021-05-02:
Step 1: Configuration Abstraction
- wrap Preferences.h (ESP32 lib) 
- define which keys exist (typing versus config.json) (in a single place, e.g. name/type/default inside a header)
- examples how to use it

Step 2a: Configuration UI
- implement functions to parse/generate data.json (POST/GET)
- implement functions to generate config.json (GET)

Step 2b: Configuration UI
- include bundle.js/css/index.html as header files
- expand wifi connect / time sync application to start the config server and show the IP

Step 3: Optional / Future
- if the wifi is not connectable -> create a wifi hotspot
- generate wifi password dynamically (once ;) ) and display on screen
- HTTPS
- password protect the configuration
- include contributors page

Simonmicro, 2021-05-02:
Okay, I finished step 01 - see here: https://github.com/Simonmicro/open-smartwatch-os/tree/feature/nvsConfig
The `config.json` should be able to use the `OSW_CONFIG_*` defines (inside an other one) to be "generated" during compile-time later on. I also created an other branch `debug/heapy` to further investigate the system crashes - for now no results.

Simonmicro, 2021-05-02:
And here are the `#define`s for the `config.json` to support the expansion of the config keys (just a quick demo program):
```c++
#include <iostream>
#include <string>

//These are the config key definitions from the osw_config.h
#define OSW_CONFIG_WTF_COLOR_R "a", "s", 179
#define OSW_CONFIG_WTF_COLOR_G "b", "s", 107
#define OSW_CONFIG_WTF_COLOR_B "c", "s", 0

//This defines a section of the json (the # allows insertion of params, even between strings)
#define MAKE_SECTION(I, T, V) "{\"label\": \"Currently not included in constant definition\", \"help\": \"Currently not included in constant definition\", \"type\": "#T", \"value\": "#V"}"
//Allow expansion of key macro into its x components (so the id, type and value)
#define SECTION_PART(X) MAKE_SECTION(X)
//This goes into the config.json - it consists of n mid parts (with a ",") and one final part
#define ALL_SECTIONS SECTION_PART(OSW_CONFIG_WTF_COLOR_R)"," SECTION_PART(OSW_CONFIG_WTF_COLOR_G)"," SECTION_PART(OSW_CONFIG_WTF_COLOR_B)

int main() {
    std::string out = ALL_SECTIONS;
    std::cout << out << std::endl;
}
```

So basically any dev just has to add a new key "at the top" (`OSW_CONFIG_`) and, if he wants to also expose it via the config ui, also append it to the define for the respective panel (`ALL_SECTIONS`).

When the ui is written and the types final, we could also add the keys name and help text to their defines - but that later on.

P.S. Now I hate macros a little bit more.

Simonmicro, 2021-05-03:
Also a quick side-note / request. Could you extend the js part of the config ui to request _two_ json files from the esp? One defining the ui (id, name, help, type and default value / html-placeholder) and the input fields, which is static build using the defines ↑ and can therefore kept inside the program code (and must not be loaded to the ram at any point) and an other json, which contains the current values (id, value)?

Simonmicro, 2021-05-05:
TODO for me (also the functions):
```
config.json:
section
label
help

data.json
like config, but no help & type
```

Simonmicro, 2021-05-06:
It is done (Step 1 & 2a), even by switching to a whole new approach, the footprint stays reasonable:
```
Before my code
RAM:   [==        ]  16.2% (used 53112 bytes from 327680 bytes)
Flash: [=======   ]  70.5% (used 1477881 bytes from 2097152 bytes)

After my code
RAM:   [==        ]  16.3% (used 53352 bytes from 327680 bytes)
Flash: [=======   ]  70.9% (used 1486797 bytes from 2097152 bytes)
```

Simonmicro, 2021-05-08:
Oh yeah, I've got still the bluetooth compangion enabled... But okay - we'll see our usages at the end :P

uvwxy, 2021-05-08:
[image]

see the develop branch. the saving step does not work yet.

uvwxy, 2021-05-08:
I'm done with part 2b, instead of the time from web app I added it as a separate app at the end. 

Anyone interested in cleaning this up a little bit, and do the part 3) ? 😅

Simonmicro, 2021-05-09:
> if the wifi is not connectable -> create a wifi hotspot

I think we should move this into a seperate wifi service, as we have now multiple apps using wifi. Also this could handle reconnections or ap creations on failures -> what do you think? If you agree, let us create an issue for that.

Simonmicro, 2021-05-09:
Also: Is there any way to create a section with just some static text? We could add there the contributor section or at least build information.

Simonmicro, 2021-05-10:
- [ ] Caching. I've noted practically every app/function, which are getting called regulary, already do this internally. Lets unify that.

uvwxy, 2021-05-18:
I added the static text and drop down. Are we done here? :)

uvwxy, 2021-05-18:
see 97dcd47

Simonmicro, 2021-05-18:
Neat, I'll just add some static version info...

Simonmicro, 2021-05-18:
> I added the static text and drop down. Are we done here? :)

Whats is the field type of static text?

uvwxy, 2021-05-22:
I think we are done here ;)

PR #92 (2021-05-19): Feature/build info
Closes #54
closed 2021-05-22 by uvwxy