Open-Smartwatch/open-smartwatch-os #55: General Conception of Apps/Menus and usage design.

xasz, 2021-05-02:
As i follewed and saw on Github there are many people including me testing around with menues and diferrent types of apps/watches and app switching.

I would love to setup a "design" language for all of us and an technical one as well.

### Current Concept:
We have Apps. Everything is a App and can to whatever it wants to.
Btn1 is the app-switch by long press and no app can change that.

### My Idea:
I would split up everything in 3 parts:

**Part 1: Launchers**
A provides the functinality to present Aps in Faces, in whatever way it chooses to. The Launcher itself can use all 3 availible Buttons to switch to Menues/Faces/Apps. A Launcher although is used to show overlays or "UI" specific things.

**Part2: Faces**
A Face can be a Watchface itself or any infomational like stepcounter/calendar/stopwatch whatever, but it is limited to exctly 1 button press. and thats chooses not by the face, but instead by the launcher. The Launcher sends this trigger to the face.
The abstract face class would also provde easy access for get Current Date/Time/Information that nobody needs to right these things over and over themself in each face.

**Part3: Apps**
A App is started by launcher and has full control over everything but also have to be able to have a "back to launcher button".

### Full Example:

**Launcher: xaszLauncher**
My Launcher decides that it toggels between Faces on Btn1 and Bt3. 
Btn2 is the trigger for Faces. Every Other Launcher could use other buttons or a combinations.

**My Face1: Watchface**
My custom or your custom Watchface what i like to have on my wrist. The Trigger Action could Start my Faverite configred App or go to the Main Menu

**My Face1: Fitness**
Shows my current Stepcounter and the triggerpress could start/stop and workout timer.

**App: Settings**
There should be a general Settings App for the launcher itself and "Display/Settings".

**App: RandomOne**
At some point i want to make a game :)

The Essential part of this is the convention of who has control over what.
If we setup a construct of abstract classes for Launcher (I am on it) and Face and App and who has the button control we could easy swap out one for  #another.
There could be many Faces which are compatible with every launcher and if you don't like the launcher but your faces, just use another one.

I have thought about the concept alot and i think we are limited with 3 buttons but i think to have well working launchers/faces/app in the community we need this kind of convention and technial abstract classes everyone can then use.

Like
osw_launcher
osw_face
osw_app

If you like the idea or the concept i would be happy to code the ground construct and pull it again.
I think it would be very nice for all of us and makes swapping apps/faces much more easier in the future.

If u have any ideas or question :) I am happy to discuss

xasz, 2021-05-03:
I would be happy take this approach on me. Making the abstrahation and Code one or Tel example/Core Launcher and Update the current Apps/faces to the nee concept If possible.

xasz, 2021-05-04:
I am working hard on a really basic but providing all functionality launcher.
I have a working copy on my branche and a small updated description in the Readme.
https://github.com/xasz/open-smartwatch-os/tree/LauncherTest

uvwxy, 2021-05-06:
Result from the discord discussion:

```c++
AppSwitcher has RTC_DATA_ATTR appPtr to remember last app/face

AppSwitcher fs = new AppSwitcher(MODE_SHORT_BTN1);
fs.addApp(new AnalogFace());
fs.addApp(new DigitalFace());

AppSwitcher launcher = new AppSwitcher(MODE_LONG_BTN1);
launcher.addApp(fs);
launcher.addApp(new StopWatch());
launcher.addApp(new Settings());
launcher.addApp(new SomethingApp());
```
[image]

closed 2021-05-11 by xasz