Open-Smartwatch/open-smartwatch-os #65: Step counter does not reset at the end of the day

rasppliances, 2021-05-06:
The step counter never resets.  It should reset every day at midnight.

Another thing about the step counter...

The target is hard coded at 360.  It would be good if we could put the default value of something more realistic like 10,000 or 5,000 in config.h.  Additionally, if we store the target in the flash memory using Preferences library, we could also write an app to adjust the goal right on the watch.

max9901, 2021-07-25:
Would it make sense to use the <Preferences.h> lib with a step-count namespace to store step values. And then write a OSWservice to deal with updating and resetting step values ?

uvwxy, 2021-07-25:
Yes 💯

max9901, 2021-07-25:
We do have a problem with reading out the time from the RTC, i was checking it tonight but we get errors (aka 0 values) when 2 cores are using the i2c bus at the same time. Should we maybe make a general system background process that actually does the time updates etc instead of giving the user full control of querying the I2c line ?

uvwxy, 2021-07-25:
Also yes :)

max9901, 2021-07-26:
[WIP]~
https://github.com/max9901/open-smartwatch-os
If time permits let me know what u think of the approach :D

max9901, 2021-07-26:
So i Have some issues with I2C on core 0. The counter reset idea is implemented where we save the last 7 days in nvs-mem.
I only want to move all I2c Sensor updates to a service on core0. However, this give very in-stable readings somehow.

uvwxy, 2021-09-05:
Recent develop branch has a step history of 7 days, and resets the first time the watch starts on a new weekday, resetting the counter and carrying over the steps to the previous day.

closed 2021-09-05 by uvwxy