Open-Smartwatch/open-smartwatch-os #66: No charging indication

rasppliances, 2021-05-06:
There's no indication of the watch being plugged in to charge.  I think we could setup an interrupt to wake the watch up when the power is plugged in, then adjust main.cpp to check what interrupt woke it up, and if it's the watch being plugged in, go to a charging animation app instead of the watch face app.  The charging app wouldn't even have to be in the cycle of apps normally shown.  The app could be as simple as the word "CHARGING" with a percentage of battery power, or it could be a battery icon that shows how much charge is there.

uvwxy, 2021-05-06:
Correct. we don't have that information hardware wise :)

(But we could guess from a value that says "not full yet")

rasppliances, 2021-05-06:
Isn't there 

`hal->getBatteryRaw()`

in power.cpp that could be used?

rasppliances, 2021-05-07:
Here's what I've found out by RTFM

- You can wake up the esp32 via a GPIO on ext0 or ext1
- You can only tell which GPIO pin woke you up on ext1
- ext1 supports GPIO32-GPIO39
- STAT_PWR that we can use to see if the charger got plugged in is on GPIO-15
- That means that the only way (that I can see) that we can get the watch to wake up when it is plugged in is to either rework the pcb so the STAT_PWR signal is on a GPIO between 32 and 39 (**not likely**), or put all other waking interrupts on ext1, then when the watch wakes up it can check what woke it on ext1, and if nothing on ext1 woke it, it must have been woken up by being plugged in

uvwxy, 2021-05-08:
What about the LED?

rasppliances, 2021-05-08:
@uvwxy I printed out the case and put the watch in that, and I can't see that LED.  We'd need a light pipe of some sort to see it, I think.

rasppliances, 2021-05-10:
@Simonmicro I'm more thinking of way to tell the watch is charging or 100% charged when it's sitting on my desk plugged in and not being used.  Having said that, I don't think an overlay is a bad idea at all.  It could even be as unobtrusive as the already existing battery/plug graphic being changed to show a battery that is charging somehow.

rasppliances, 2021-05-10:
@Simonmicro Another thought.  I was originally thinking that it would be activated when plugged in while not "on".  But, as I found, that interrupt can't be determined when the watch wakes up, so my solution above was a work around for that limitation.

Simonmicro, 2021-05-12:
Hmm, I'll further participate in this, when I finish up my other ideas & tasks. Just letting you know, I'm happy to further work on this, but not right now :P

uvwxy, 2021-05-16:
The charging indication is covered by a hardware LED. 
If you come up with a sw solution that manages to correctly indicate this, feel free to create a pull request.
will close the issue for now :)

closed 2021-05-16 by uvwxy