Open-Smartwatch/open-smartwatch-os #77: WiFi service (and reworking the services)

Simonmicro, 2021-05-10:
- [ ] when the wifi service exists (and it creates an ap on missing connection) we can also remove any need to preconfigure the config.h and make its use optionally
- [ ] auto-ntp (when connected to wifi, auto-pull ntp; should be integrated into the service ↑)
- [ ] if the wifi is not connectable -> create a wifi hotspot
- [ ] generate wifi password dynamically (once ;) ) and display on screen

This TODO came up while working on #54

Simonmicro, 2021-05-17:
@SeanReg I know you are the original author, so I'm letting you know:

I plan to introduce to the services a similar header behavior like with the `OswConfigKeys`. So we have some kind of namespace to access all the services (so any app could later on access & interact with them) and they are instanced as needed (only if `#defined` an instance of them get created).

Are you OK with this or do you have something else in mind? I have to admit that some kind of singleton design could also be utilized, but I have not thought about that further now...

SeanReg, 2021-05-17:
Right now the Services have a singleton design via the ServiceManager that allows you to do the following:

`ServiceManager::getInstance().registerService`
`ServiceManager::getInstance().getService(Services::<SERVICE_NAME>)`

You can #ifdef around each of the `registerService` to prevent the service from being registered. Thinking maybe this will keep services a bit more self contained?

Simonmicro, 2021-05-20:
Yeah, I've seen this - but I assume your `registerService` call works dynamically and therefore on the heap? I would like to move it into the stack/program code like I've already done with the config keys. Basically you define a static array, containing all the pointers to the statically created instances (maybe contained inside a namespace, so any app could access it directly [hey - also no search & return overhead anymore]). Yes, this removes the ability to "register"/"unregister" any services, but t.b.h. do we really need this during runtime? And if we _really_ need to disable a service at runtime, we could just add an `enabled` boolean to skip its `loop()`.

TL;DR Your current system is fine, but I would like to make it a little bit more static and remove some overheads.

Am I overlooking anything or do you have any further doubts? Otherwise I'll proceed with this idea now...

Simonmicro, 2021-05-20:
@SeanReg Here is my current progress regarding the "new" service structure - feedback is (as always) welcome:
https://github.com/Simonmicro/open-smartwatch-os/tree/feature/wifiService

Simonmicro, 2021-05-20:
## Current TODOs

- [ ] rework servicemanager to be more static
- [ ] clean up main.cpp and strip out the task stuff for core 0
- [ ] move gps into own service
- [ ] break `config.h` defines again
- [ ] create heap size logger

uvwxy, 2021-05-22:
@Simonmicro I looked at your service structure refactoring and would like to merge it as soon as you are done with it. your pull request was still marked as draft.

Simonmicro, 2021-05-31:
# service specs / implementation todo

- [ ] wifi service key `autoenableandconnectWifi`
- [ ] connect/disconnect (works with ssid/pass  OswConfigKeys)
- [ ] loop, setup, stop
- [ ] getNativeHandler(); maybe getVisibleWiFiAPs(), getClientIP(), getStationIP()
- [ ] enableAP, disableAP
- [ ] enableWiFi, disableWiFi
- [ ] enableAutoNTP, disableAutoNTP (controlled with OswConfigKeys)
- [ ] isConnected(), isWiFiEnabled(), isAPEnabled(): true for wifi or always true if ap

closed 2021-06-02 by Simonmicro