Open-Smartwatch/open-smartwatch-os #79: Reset gfx on App change

xasz, 2021-05-11:
I run into some problems with apps on messed up positionioning, font, size, color.
After a ton of stupid testing i figured that my code works fine, but if i change the app and come back to mine one, some stuff like alignment and font size got messy.
We should add an gfx->resetToDefault() and the appswitcher should call that before an app switch happens, so every app has the same starting point.
Otherwise for example if u don't set the font size and use the "default" size, and a other app does, it really can get messed up :)

rasppliances, 2021-05-12:
I think that a two prong approach would be best to this.  First, as you said, reset the graphics, but only if that doesn't perform a performance hit.  Second, make it a best practice to developers that nothing is set to the default and that they should make a habit of setting whatever values they need/expect before executing their code.  Using your font size as an example, best practice would be to always set the font size first, assuming it's not the default.

This is the kind of assumption I make when I'm "coding" for my job (I put that in quotes because I do use actual JavaScript code in my work, but it's just within one product mostly using objects specific to that product that have no use in the outside world.  I'm not a coder, really.)

xasz, 2021-05-12:
I really do agree only to a small part.
I think it is best practive and well, it is the need, for the developer to set his settings for size, font and whatever.
But u cannot tell the developer he always have to reset all the settings manual. Which for first: Does require alot of copy and paste code from one app to another and second it is not upgrade compatible.

Lets say you have some Text Methods like:
```
setSize(..)
setFont(..)
setAlignment(..)
```
Now we tell everyone, every app have to make sure it sets this 3 options to its liking and then run for example print(...).
When everybody does that this works fine, until the point there, that there is a new fomatting option like for example:

`setLineSpacing(..)`

Now when a new app uses this method, u break all the other formatting without touching each existing app and set the setLineSpacing to the apps liking :) There really should be a default state, that a app can assume.

uvwxy, 2021-05-13:
I'm working on this

max9901, 2021-05-16:
I'm a fan of adding a reset function into the library we can call on watch face switch. (And optionally let this restore to settable default values so developers can use a certain style across all there faces.)

If u guys agree i can implement it.

uvwxy, 2021-05-16:
I fixed this with abb2080 and 313d413

closed 2021-05-16 by uvwxy