Open-Smartwatch/open-smartwatch-os #103: Browser config does not keep 24 hours setting

rasppliances, 2021-05-25:
When I use the browser config and uncheck the Use 24h time format? option, then save the settings, this setting is not saved.

Steps to reproduce

1. Go to the Configuration UI app and click Connect
2. Use your browser to login in using the settings displayed on the watch.
3. Scroll to the bottom and uncheck Use 24h time format?
4. Click Save
5. Note the Saved message
6. Reload the page
7. Note that Use 24h time format? is checked again.

airtonzanon, 2021-05-25:
I took a look on the developer branch and there's the same problem but with the colors (section Theme & UI).

uvwxy, 2021-05-26:
Could you check this again. I have changed the keys of settings recently, which will reset to default values once to reinitialise the config partition.

... if we can rule this out, I can check this again

airtonzanon, 2021-05-28:
@uvwxy I just rebased with the last version of `develop` and it still happening.

When I take a look on logs I just see that it writes until `foreground color (dimmed)`

```
Going to write config key id a as SSID...
...done!
Going to write config key id b as Password...
...done!
Going to write config key id s2 as Display Timeout...
...done!
Going to write config key id s1 as Display Brightness...
...done!
Going to write config key id s3 as Display Overlays...
...done!
Going to write config key id s4 as Display Watchface Overlays...
...done!
Going to write config key id s5 as Raise/Tilt to Wake...
...done!
Going to write config key id s6 as Raise to Wake Sensitivity...
...done!
Going to write config key id s8 as Tap to Wake...
...done!
Going to write config key id s7 as Light Sleep...
...done!
Going to write config key id e as Date format...
...done!
Going to write config key id f as Daylight offset...
...done!
Going to write config key id h as Timezone...
...done!
Going to write config key id g as Use 24h time format?...
...done!
Going to write config key id c1 as Background color...
...done!
Going to write config key id c8 as Background color (dimmed)...
...done!
Going to write config key id c2 as Foreground color...
...done!
Going to write config key id c9 as Foreground color (dimmed)...
...done!
```

I tried to change the `Primary Color` for example, and it won't keep the configuration that I changed.

rasppliances, 2021-05-31:
I tried it with the latest build in an incognito window.  All the colors saved no problem, but the time/date information did not.  24 hour, daylight offset, time zone and date format would not save.  Even after I changed them while changing a color as well.  The color would change and save, but the time/date setting would not.

Simonmicro, 2021-05-31:
CONFIRMED BUG. Will be fixed asap (we were running out of stack space while parsing the json with the data).

Simonmicro, 2021-05-31:
Will be fixed in PR #110 

Feel free to reopen this, if the problem persists after merging...

rasppliances, 2021-05-31:
Not reopening, but asking, should issues be closed before they are fixed?  In my line of work (I implement ITSM/help desk tools) incidents aren't closed until they are fixed or a workaround is given.  Granted, this is targeted to be fixed by a current pull, but it's not fixed yet.  Just my thoughts.

Simonmicro, 2021-05-31:
Okay, very fair point. Consider this issue open and closed at the same time :laughing:

airtonzanon, 2021-06-03:
Just tested! Working well on develop! Thanks @Simonmicro

closed 2021-06-03 by Simonmicro

rasppliances, 2021-06-21:
It looks like the pull has been merged, and I'm still experiencing the same problem.  I cannot turn off 24 hour time on my watch.  There used to be a setting for it in config.h. What happened to that, and why was it taken away?

airtonzanon, 2021-06-22:
@rasppliances it was merged to the branch develop not master.

rasppliances, 2021-06-22:
@airtonzanon Thanks!  That explains it.  What about putting these settings back in config.h?  Is there any way to do that in the meantime?  I looked, but couldn't figure it out.

Simonmicro, 2021-06-23:
Hmm, our goal was to move away from the compile-time configuration, but I see no problem with keeping the defaults inside the `config.h`. I mainly removed these settings to force developers to use the new config system (and so that nobody would wonder why some apps would just not respect the setting from the web ui).

rasppliances, 2021-06-23:
That makes sense.  Moving away from compile time settings would make the watch more accessible to people.  They could use the watch without having to install a toolchain they don't use for anything else.

Is it possible for someone to choose to "add back" a configuration into config.h without any other change being made to the code?  Alternatively, where can I go in the code to change it just for myself?  Kind of like how I can go into main.cpp and comment out the snake game so it doesn't appear on my watch.  Where can I go to make it so my watch defaults to 12 hours?

Simonmicro, 2021-06-23:
Take a look into `src/osw_config_keys.cpp`. All the settings and their defaults are defined here. Note that you also have to modify `configVersionValue` inside `include/osw_config.h` to force the watch to re-apply the default values (just set an other number).

If you wish to have a default `config.h` again, just create an issue, so we could track & discuss this there further.

rasppliances, 2021-06-23:
@Simonmicro  This issue goes away soon, so I don't want to change the design plan for less reliance on config.h.  I'll look into the files you mentioned and see what I can work out.  Thanks for your help!

commit f3e89d6f (2021-06-01): Fixed #103