Open-Smartwatch/open-smartwatch-os #104: Download center for watchfaces, 3D cases and more

mat942, 2021-05-25:
Hello all,

Given user base is growing by the day and OSW offers a very high degree of customization I think a place to showcase each user creativity could be great. The idea would be to have a place we can download watchfaces and integrate them easily (maybe redo code so its a single file to replace?). It could also be used for 3d cases or apps later on. Something like an "OSW store" but all free with credit to creators of course.

Any thoughts? Happy to assist

richtepa, 2021-06-18:
I would like to help too. Is there a plan?
Do we make watchfaces in a single file first?
Do we want to use an available system for sharing/downloading or do we want to code it ourseves?
The user needs to compile and program the whole firmware. OSW-OS is hosted on GH. Is there a good way to integrate the Download center on GH (another Repo in combination with GH-Pages)?

richtepa, 2021-06-18:
On a download center we could also make it possible to configure and install the FW on the watch via USB (later OTA).
By selecting apps/watch-faces we could change the code correspondingly. Uploading should not be a problem, too. We only need to have an idea on how to build the FW from the code (client or server).

mat942, 2021-06-18:
I was thinking of runnign a wordpress site than can handle upload dowload and have either a unified watchface format or store firmware elfs.

mat942, 2021-06-18:
> I would like to help too. Is there a plan?
> Do we make watchfaces in a single file first?
> Do we want to use an available system for sharing/downloading or do we want to code it ourseves?
> The user [...]

I think the easiest way is make watchfaces single file and shareable first

richtepa, 2021-06-18:
> I was thinking of runnign a wordpress site that can handle upload dowload

Wordpress is a simple solution, too. I'm not very experienced using Wordpress but there are a lot of plugins. Maybe there is the perfect one out there.

> have either a unified watchface format or store firmware elfs

I'm not sure if elfs are the way to go. If you provide a watchface as .elf it will be the whole OS. This way you cannot add apps or change the config. Or did I miss something?

> I think the easiest way is make watch faces single file and shareable first

Yes. :100:

> We can then work on a peer review or voting system to score the best in a gallery format with easy implementation on the user side.

For watchfaces e.g. upload
* Watchface itself
* Screenshot / Image (maybe provide easy tool to create these screenshots)
* Title + (short) description

Voting system is a nice idea.

> Would be good to keep the watch face version independent and build independent (light/gps)

I'd prefer light+gps by default, but gps-only should be possible, too (e.g. if someone wants to shows a map or uses any other feature)

mat942, 2021-06-18:
something like this could work https://wordpress.org/plugins/bbp-voting/

mat942, 2021-06-18:
fully agree with your vision @richtepa

richtepa, 2021-06-18:
Is @uvwxy d'accord with our plan?

Then I'll search and test for suitable wordpress plugins this weekend.

richtepa, 2021-06-18:
@mat942 is there something you want to do? (I don't want to take it away from you 😉 )

Simonmicro, 2021-06-18:
> I think the easiest way is make watch faces single file and shareable first

:100:  - Problem is we are unable to compile a stand-alone add-on for the firmware, as the latter one may not include all used functions (optimizations, optimizations).
- @SeanReg build a LUA integration and we could try to port existing apps and extend the API to make it really usable (may the best way)!
- The better way (I think) would be to utilize something like a website / docker container to compile the OS with all needed add-ons directly for the watch and then sent it over (OTA is WIP and will be soon :tm: finished)! A cool example would be https://github.com/benzino77/tasmocompiler
- Additionally everyone should be able to host a copy of the repo by himself easily? But that's just a personal preference...

> I was thinking of runnign a wordpress site that can handle upload dowload and have either a unified watchface format or store firmware elfs.

Brrr - who is up for maintaining that site in a secure way? Not me... I'm thinking a more static approach - like Hugo and then integrating into GitHub for the file / source storage (take a look into https://cinnamon-spices.linuxmint.com/ and his Github Repo https://github.com/linuxmint/cinnamon-spices-applets).

But lets see, what you guys can build!

mat942, 2021-06-18:
We might want to create a community branch to have an architecture development for shareable watch faces

mat942, 2021-06-18:
good catch on the web maintenance, we will need to figure out a system for users to submit their ideas though so we keep the site populated

mat942, 2021-06-18:
your lead with tasmo also looks very promising

Simonmicro, 2021-06-18:
> something like this could work https://wordpress.org/plugins/bbp-voting/

BUT this looks very good for a feature request page?! Maybe, maybe... Someone interested?

mat942, 2021-06-18:
> > something like this could work https://wordpress.org/plugins/bbp-voting/
> 
> BUT this looks very good for a feature request page?! Maybe, maybe... Someone interested?

https://wordpress.org/plugins/simple-feature-requests/

richtepa, 2021-06-18:
> The better way (I think) would be to utilize something like a website / docker container to compile the OS with all needed add-ons directly for the watch and then sent it over (OTA is WIP and will be soon ™️ finished)! A cool example would be https://github.com/benzino77/tasmocompiler

That's my favorite way, too. Creating the compiler will be hard.

> Brrr - who is up for maintaining that site in a secure way? Not me... I'm thinking a more static approach - like Hugo and then integrating into GitHub for the file / source storage (take a look into https://cinnamon-spices.linuxmint.com/ and his Github Repo https://github.com/linuxmint/cinnamon-spices-applets). Additionally everyone should be able to host a copy of the repo by himself easily? But that's just a personal preference...

Thats why I suggested the GH approach earlier. We could create some small scripts and only display/provide the watchfaces from GH on the external download center. Editing would be in GH. (like Hugo but more tailored to our needs)

> But lets see, what you guys can build!

I think we are still in the idea/planning phase. The more people share their ideas and taught on the different methods, the better the result will be.

richtepa, 2021-06-20:
I have no idea on how to combine the files into one.

This weekend I started writing some code (https://github.com/richtepa/osw-store) for the download center. It's a bit rough and has a lot of loose ends I need to fix. (+cleaning the code up, security, more features, etc.) The design is also not pretty 😉 .

A preview of this possible solution can be found here: https://osw.richter.dev
If you don't want to "sign up":
username: demo
password: 1234

If the feedback is mostly positive I will keep working on this way.
ANY feedback is desired, negative too!

uvwxy, 2021-06-20:
Sorry for the late response. Thank you for the great discussion!
Here is what I picked up + my two cents:

- first share the **c++ files** until the lua script part is ready (it's missing gfx at the moment and uses ArduinoGraphics instead). Once the lua part is worked out, it should be easy to port to c++, and make something dynamic to download them onto the watch from the watch (think: far in the future ;) )
- I wrote [OswServiceTaskRawScreenServer.cpp](https://github.com/Open-Smartwatch/open-smartwatch-os/blob/develop/src/services/OswServiceTaskRawScreenServer.cpp), to **grab raw rgb888 screen frames directly form the watch** to create screenshots while apps are running (convert to png on desktop), this way we could create nice gifs with a nice bash loop. even if the gfx code can be compiled on desktop, I guess most will work with developing directly on the watch, and hence the idea to grab the screens from there. The server is not working yet, but the general code/stubs are there
- static vs wordpress: personally I prefer **static**, generated from markdown (maintained via github). either mkdocs (like the project page) or jekyll (like my website p3dt.net, there even nice tools to always show the latest additions)
- creating customized builds via docker: this is something I could do, I have a server that can run docker. the basic idea would be to use flags to enable/disable apps and then createa platformio profile to run the compiler in the container

richtepa, 2021-06-25:
> [...] ANY feedback is desired [...]

No feedback is feedback, too.

> I wrote OswServiceTaskRawScreenServer.cpp [...] The server is not working yet, but the general code/stubs are there
> [...] creating customized builds via docker: this is something I could do

I think this is a point we could add to the backlog. But if this is something we (you) can do, great 👍 .

> first share the c++ files [...]

Yes, in my opinion the page itself (providing the c++ files) would be the first step. 

> static vs wordpress [...] generated from markdown (maintained via github). either mkdocs (like the project page) or jekyll (like my website p3dt.net, there even nice tools to always show the latest additions)

I am not familiar with these tools. Is there a way to add a voting system (if we like to)? If there is nobody else, I think this task will be yours, too 😉 .

mat942, 2022-04-14:
Hi guys! discussion on this has been pretty much frozen since summer. Has anybody advanced any steps?