Open-Smartwatch/open-smartwatch-os #116: Rise to wake not working

rasppliances, 2021-06-20:
I have configured the rise to wake threshold as low as 0 and as high as 255, and the watch does not wake when I bring up my arm and twist my wrist.

uvwxy, 2021-06-20:
I had the same issue. thanks for creating a ticket to keep an eye on this.

mpegg4, 2021-07-22:
Looks like https://github.com/Open-Smartwatch/open-smartwatch-os/issues/119 was related, fixed & closed.  Maybe this ticket is too?

AvdHdn, 2021-08-17:
I noticed that "tap to wake" does not work on my watch when I manually put it to deep sleep mode.
When the watch would time out by itself (on the clock watchface) the tap to wake function does work.
Do you have the same issue?

uvwxy, 2021-08-17:
> I noticed that "tap to wake" does not work on my watch when I manually put it to deep sleep mode.
> When the watch would time out by itself (on the clock watchface) the tap to wake function does work.
> Do you have the same issue?

This is intentional: https://github.com/Open-Smartwatch/open-smartwatch-os/blob/master/src/apps/main/switcher.cpp#L44

closed 2021-12-07 by uvwxy