Open-Smartwatch/open-smartwatch-os #119: Don't allow setting screen timeout to 0 seconds

exponentactivity, 2021-06-29:
Setting screen timeout to 0 seconds puts the screen to sleep constantly/instantly, locking the user out.

I tried disabling screen timeout by setting it to 0 due to #116 but instead i locked myself out and can't figure out how to reset the timeout to a usable value.

Instantly putting the display to sleep/turning it off doesn't seem to serve a useful purpose. Could instead disable the timeout?

exponentactivity, 2021-06-29:
In case someone else ends up in this situation:

Editing `./include/osw_config.h` and changing `configVersionValue` to a new value (i set it to 3) will reset the config and make the watch usable again.

Thank you to the developers for commenting the code!

Simonmicro, 2021-06-30:
I guess a check to disable the timeout on zero would be a good idea.

Simonmicro, 2021-07-08:
Should be fixed - could you confirm this with a fresh build from the `develop` branch?

exponentactivity, 2021-07-09:
Now the watch crashes when changing apps, before i can confirm the fix.

After uploading the new `develop` firmware to the watch and rebooting, the watch boots like expected. 
I change app, trying to get to the web-ui setup to confirm the fix, but when i try to change the app from water_level the watch crashes and the monitor terminal outputs `cannot open /data/lua/apps/stopwatch.lua: No such file or directory`

Full output from booting the watch until it crashes (as i don't trust myself to know what else might be relevant):

```
include/osw_config.h: Accessing key id j
include/osw_config.h: Accessing key id k
include/osw_config.h: Accessing key id l
include/osw_config.h: Accessing key id s2
include/osw_config.h: Accessing key id s1
include/osw_config.h: Accessing key id s3
include/osw_config.h: Accessing key id s4
include/osw_config.h: Accessing key id s5
include/osw_config.h: Accessing key id s6
include/osw_config.h: Accessing key id s8
include/osw_config.h: Accessing key id s7
include/osw_config.h: Accessing key id f
[E][Preferences.cpp:492] getBytesLength(): nvs_get_blob len fail: f NOT_FOUND
h
include/osw_config.h: Accessing key id g
include/osw_config.h: Accessing key id c1
include/osw_config.h: Accessing key id c8
include/osw_config.h: Accessing key id c2
include/osw_config.h: Accessing key id c9
include/osw_config.h: Accessing key id c3
include/osw_config.h: Accessing key id c4
include/osw_config.h: Accessing key id c5
include/osw_config.h: Accessing key id c6
include/osw_config.h: Accessing key id c7
Config loaded! Version? 3
Boot count? 24
[E][esp32-hal-cpu.c:115] removeApbChangeCallback(): not found func=40113B40 arg=3FFBEDF8
Setting brightness to 128
src/services/OswServiceTaskWiFi.cpp: [Mode] Off
Setup Done
========= Memory Monitor =========
core 0 (high):  4096B of 4096B
core 1 (high):  1760B of 8192B
heap (high):    302496B of 302496B
heap (curr):    160176B of 302496B
src/services/OswServiceManager.cpp: Background worker started.
========= Memory Monitor =========
core 0 (high):  632B of 4096B
core 1 (high):  1760B of 8192B
heap (high):    222064B of 302496B
heap (curr):    160176B of 302496B
========= Memory Monitor =========
core 0 (high):  756B of 4096B
core 1 (high):  1760B of 8192B
heap (high):    222064B of 302496B
heap (curr):    160176B of 302496B
========= Memory Monitor =========
core 0 (high):  756B of 4096B
core 1 (high):  1760B of 8192B
heap (high):    224164B of 302304B
heap (curr):    163252B of 301568B
========= Memory Monitor =========
core 0 (high):  756B of 4096B
core 1 (high):  1760B of 8192B
heap (high):    238800B of 298492B
heap (curr):    178600B of 297776B
========= Memory Monitor =========
core 0 (high):  756B of 4096B
core 1 (high):  1760B of 8192B
heap (high):    254704B of 295132B
heap (curr):    232380B of 294468B
========= Memory Monitor =========
core 0 (high):  756B of 4096B
core 1 (high):  1760B of 8192B
heap (high):    260820B of cannot open /data/lua/apps/stopwatch.lua: No such file or directory
293776B
heap (curr):    188984B of 297760B
========= Memory Monitor =========
core 0 (high):  756B of 4096B
core 1 (high):  2340B of 8192B
heap (high):    ========= Memory Monitor =========
core 0 (high):  868B of 4096B
core 1 (high):  2340B of 8192B
heap (high):    269632B of 302420B
heap (curr):    160512B of 302420B
269632B of 302420B
heap (curr):    160532B of 302404B
```

Please, let me know if you need anything else or if i should make a seperate issue for this.

Simonmicro, 2021-07-09:
This is a separate issue - it seems you tried to compile a lua app into the os and failed to add the script itself?! This (should) not be related to this issue. Try reverting the main.cpp to the original state of the develop branch, please.

exponentactivity, 2021-07-09:
The feature works as intended. Thank you.

closed 2021-07-09 by exponentactivity