Open-Smartwatch/open-smartwatch-os #120: Digital clock face - text background incorrect after updating color settings using Configuration UI

mpegg4, 2021-06-30:
Thanks to brrMoment for surfacing this and the collab to identify the fix. 
When updating foreground/background color using Configuration UI, digital watchface (and some other apps) draw text without deriving ui background/foreground color and applying. This creates issue for first time users who are not familiar with coding.

Example digital watchface screenshot with background color issue:
[image]

Other switcher apps have this adjustment made:

`hal->gfx()->setTextColor(ui->getPrimaryColor(), ui->getBackgroundColor());`

Suggested fix - update watchface_digital.cpp and add the same in either `void OswAppWatchfaceDigital::setup` or `void OswAppWatchfaceDigital::loop`:

`hal->gfx()->setTextColor(ui->getPrimaryColor(), ui->getBackgroundColor());`

Test of digital clock after applying:

[image]

Simonmicro, 2021-07-08:
Should be fixed - could you confirm this with a fresh build from the `develop` branch?

mpegg4, 2021-07-08:
@Simonmicro thanks - will check it out and feedback here and in discord. Awesome work!

mpegg4, 2021-07-08:
@Simonmicro Ack - looks great now!

commit 75030c9c (2021-07-08): Fixed #120 (reload ui colors on save instantly & updated apps to use ui helper function)
commit 83fdb73a (2021-07-08): Merge pull request #128 from Simonmicro/develop

Fixed #120 (reload ui colors on save instantly & updated apps to use …
closed 2021-07-08 by mpegg4