Open-Smartwatch/open-smartwatch-os #121: Any Java/Kotlin API to interact with?

maifeeulasad, 2021-07-02:
If I want to fetch data from this smartwatch, then how can I do it? I was looking for something in the doc, but couldn't find it.

Simonmicro, 2021-07-02:
No there is currently no API nor is one planned - what data do you need?

maifeeulasad, 2021-07-02:
I'm planning to buy one and then make some applications to interact with this smartwatch. 

Actually, I want bi-directional communication.

I'm still reading the OS source, then I may get to know, how can I modify the OS to create a custom server and etc. But I'm quite confident about the Android part. I can easily connect Android to almost anything.

Simonmicro, 2021-07-02:
That sounds nice, I would recommend you to create a service task inside the osw-os and to connect the watch to your smartphone using Bluetooth. This ensures the watch also works when the user is not at home.

Please note that our current hardware revision has some serious limitations, greatly impacting our ability to use Bluetooth without crashing (and also using nearly all memory up). I would therefore recommend you to wait for the next hardware release until you start developing any Bluetooth networking stuff (ofc you could just compile the os yourself and strip out all the wifi stuff - this should give you enough room for now).

Simonmicro, 2021-07-15:
How about we work on that together - I'm happy to implement the C++/BLE part!

Question is what data you need to set/get and which way of communication you want. WiFi works most of the time - BLE must be further investigated (maybe for now only compile one at a time)...

Do we need periodic wake ups? Long data retention? Ideas? :grinning:

uvwxy, 2021-07-21:
@maifeeulasad let's do further discussions on discord. If your question in this issue has not been answer re-open  ;)

closed 2021-07-21 by uvwxy