Open-Smartwatch/open-smartwatch-os #122: stopwatch bug

fafa-junhe, 2021-07-02:
I found a bug when I click the reset and continue buttons of the stopwatch at the same time, the time will increase unexpectedly.I found a bug when I click the reset and continue buttons of the stopwatch at the same time, the time will increase 
 unexpectedly.
```c++
  if (hal->btnHasGoneDown(BUTTON_3)) {
    if (reset) {  // Start
      start = hal->getLocalTime();
    } else {  // Continue
      sumPaused += diff;
      start = hal->getLocalTime();
    }
    running = true;
    reset = false;
  }

  if (hal->btnHasGoneDown(BUTTON_2)) {
    if (running) {  // Stop
      running = false;
    } else {  // Reset
      diff = 0;
      sumPaused = 0;
      reset = true;
    }
  }
```

Simonmicro, 2021-07-08:
Should be fixed - could you confirm this with a fresh build from the `develop` branch?

fafa-junhe, 2021-07-09:
Yes - It fixes perfectly.

commit 77067749 (2021-07-08): Fixed #122 (time jumps while holding stop&continue)
commit dfda094d (2021-07-08): Merge pull request #127 from Simonmicro/develop

Fixed #122 (time jumps while holding stop&continue)
closed 2021-07-09 by fafa-junhe