Open-Smartwatch/open-smartwatch-os #124: Latest develop branch results in black screen.

Kirottu, 2021-07-08:
When i try to run the latest development branch version I only get a black screen on the watch.

Simonmicro, 2021-07-08:
What is the output on serial? Possibly a boot-loop, but without further information we can't debug this.

Kirottu, 2021-07-08:
How do i monitor the serial connection?

Simonmicro, 2021-07-08:
Somehow you uploaded the firmware - right? Use Visual Studio Code also for this (click at the bottom left on the plug-symbol).

Kirottu, 2021-07-08:
I use the command line and PlatformIO core. I would like to use that if it's possible.

Simonmicro, 2021-07-08:
Okay, the OS is probably normally booted - did your model every work? Did you made sure to compile the correct edition for your model (like using light for the aliepress ones, gps for the one with gps and so on)?

Idea: Something went wrong with the color setup - please modify line 35 (`  const short configVersionValue = 2;    // Change this when you want to clean the config on next boot` in file `include/osw_config.h` to reset any remaining system config to their defaults.

Kirottu, 2021-07-08:
The watch does work, I have it in use right now with the current master branch os, and I compiled it with the same command as the master branch. I will try out that idea when I get home.

Kirottu, 2021-07-14:
Any new ideas on how to try to make this work?

Simonmicro, 2021-07-14:
I have no idea what's going on - everything seems to boot fine. Could I send you a known good build of the develop branch? Maybe something goes wrong during compiling...

Kirottu, 2021-07-14:
I can definetly try that out. If that doesn't wrong then it's really weird.

mpegg4, 2021-07-14:
I had some issues with configs in the past,  maybe can try VSCode -> Erase Flash to ensure the watch is fully clean before fresh upload and monitor?
I just tried fresh develop branch seemed ok on win platform.

[image]

Simonmicro, 2021-07-15:
> I can definetly try that out. If that doesn't wrong then it's really weird.

Okay, this one should work.

[firmware.zip](https://github.com/Open-Smartwatch/open-smartwatch-os/files/6822238/firmware.zip)

Kirottu, 2021-07-15:
How do i flash that to the OSW? (with cli tools). I also tested erasing the OSW and then reflashing it with the latest develop branch which didn't work.

Simonmicro, 2021-07-15:
Put it into the `.pio/build/.../` folder (`...` for your compiled edition) and replace the existing `firmware.bin`. Then you just have to reupload - make sure to not recompile.

Kirottu, 2021-07-15:
That works completely fine now, and it is the latest development version. So something goes wrong with the build then.

Simonmicro, 2021-07-15:
Okay... 
- Run `git status` and look for any changed files.
- Pull newest develop `git pull origin develop`.
- Make sure you are on develop `git reset --hard origin/develop`.
- Clean everything `git clean -f`.
- Use the default `config.h.example` as `config.h`.
Try it again.

Kirottu, 2021-07-16:
Thats interesting, it works out now. I'll try to spot the config change which caused this.

uvwxy, 2021-07-21:
Can we close this?

Simonmicro, 2021-07-21:
I guess yes. @Kirottu feel free to post your finding when ever you are ready - but for now this issue is finished / solved.

closed 2021-07-21 by Simonmicro