Open-Smartwatch/open-smartwatch-os #131: on sleep - button presses should wake up watch

mpegg4, 2021-07-14:
Not an issue but a feature request - would be nice to wake up the watch face by pressing any button (ie not have to tap screen). Tap to wake as only wake mechanism is not always obvious.

mpegg4, 2021-07-14:
Maybe can introduce a configuration line item that can be turned _off_ but is defaulted to _true_, and pass it into these sleep functions (which are currently hardcoded):

power.cpp:

`void OswHal::lightSleep() {
  _isLightSleep = true;
  doSleep(this, >>>configDrivenButtonWake<<);
}

void OswHal::lightSleep(long millis) {
  _isLightSleep = true;
  doSleep(this, false, >configDrivenButtonWake<<,millis);
}`

config.h:

`#define >configDrivenButtonWake<< 1 `

something like that

airtonzanon, 2021-07-15:
This is already possible if you disable the tap to wake.
You can use the boot button to wake the watch, @uvwxy introduced this a while ago.

uvwxy, 2021-07-21:
@mpegg4 thank you for your contribution. The intended "button press should wakeup watch" happened when you deselect both tilt and tap to wake. 

I can see that this was not explicitly clear though. This is handled in `power.cpp` as follows:

```c++
if (wakeFromButtonOnly || // force button wakeup
      (!OswConfigAllKeys::raiseToWakeEnabled.get() && !OswConfigAllKeys::tapToWakeEnabled.get())) {
    // ore set Button1 wakeup if no sensor wakeups registered
    esp_sleep_enable_ext0_wakeup(GPIO_NUM_0 /* BTN_0 */, LOW);
  } else if (OswConfigAllKeys::raiseToWakeEnabled.get() || OswConfigAllKeys::tapToWakeEnabled.get()) {
    // tilt to wake and tap interrupts are configured in sensors.setup.
    esp_sleep_enable_ext0_wakeup(GPIO_NUM_34 /* BMA_INT_1 */, HIGH);
  }
```

I have differentiated the two ways to trigger sleep mode:
- autom. sleep on the watch screens: uses the settings (wakeup by either tilt, tap, or if neither: button1)
- sleep by long-pressing the button1: wakeup by button1

Now the second case also uses the settings.

I personally preferred to have the watch faces wakeup by tap/tilt, and use the manual sleep entry to keep the watches asleep during shipping.

With the new option we can select tilt or tap _and_ button1 in the settings. Now this means that even if only the button is used as a wakeup source, the accelerometer remains active and fires interrupts (which are ignored by the esp32). Even if the BMA400 is not drawing _that_ much power, the final configuration is not optimal.

The reason why the wakeup sources currently are accelerometer _or_ button, is because only one button can be explicitly registered as a wakeup source. if multiple pins should be the wakeup source, they all must work in the same fashin, either all active high, or low (and that is not the case with half the buttons pulled high, and the other low)

... i will definitely look into this to improve the wakeup sources

mpegg4, 2021-07-21:
Thanks @Simonmicro  @uvwxy & @airtonzanon  (sorry @airtonzanon missed your notification) ! 

Definitely problematic if buttons different response or wakeup event model will have conflicts or double fires! I also appreciate the elevated education and conversation level here on my first code contribution to any project ever - thank you.

For context - came across this scenario when I had finger on button and watch went to sleep - pressing button to wake up the screen and continue the experience was what prompted the PR . Unfortunately I fall into camp of less-then-bright people who did not intuitively tap/tilt to wake up or figure out there's a specific button to wake up.

I am OK to close this ticket & rollback change if that's whats best for the project - no issue at all.

mpegg4, 2021-07-22:
Closing issue - support for functionality is seeded and now there are defaults introduced that can be overridden for individual preferences.

closed 2021-07-22 by mpegg4