Open-Smartwatch/open-smartwatch-os #141: Open-smartwatch application for Smartphone

RuffaloLavoisier, 2021-08-04:
Hi, I got a lot of inspiration from my watch.  You can do more various things by linking your wristwatch and smartphone with ESP32. I have a similar function.

[image]

# 1. Notification
It informs the alarm that comes to the smartphone with a clock.

# 2. Device Finder
Default : If they are out of a certain distance from each other, the wristwatch gives an alarm to inform the user of the risk of loss. (This is usually the case when you are looking for a smartphone. )

## from phone
When a finder function is activated on a smartphone, the location of the clock can be checked through an alarm built into the clock. 

## from watch
Select the mode from the clock and make the smartphone ring to locate it. 

Is there anyone who wants to contribute to this?

kekko7072, 2022-06-09:
Hey I am a mobile app developer with flutter and I work full stack (server + front end), but I am not very skilled at electronics but I want to improve. I had your same idea, and it is exiting. I've abandoned the idea because my skill using and editing the firmware of the watch were very poor, but if you help me with the hardware/software of the watch I will be glad to develop the app. Last year I bought a watch light edition but I was not able to edit the firmware and run bluetooth.

RuffaloLavoisier, 2022-06-09:
@kekko7072 
Hi, there
Is it possible to build OSW firmware now? Although there is not enough memory to use Bluetooth, it has been confirmed that it can be implemented. Personally, if you have a prototype, the pull request is always open.Please feel free to open a discussion

kekko7072, 2022-06-10:
Hey @RuffaloLavoisier what do you mean for OSW?

RuffaloLavoisier, 2022-06-10:
Sorry, `Open-smartwatch` build !

simonmicro, 2022-06-10:
-> https://github.com/Open-Smartwatch/open-smartwatch-os/blob/bc08af49ac33bde2c81f463ee0a1be0812b8d388/src/services/OswServiceTaskWebserver.cpp#L291

PierreBresson, 2024-10-15:
Hi,

I've created a prototype for a smartwatch app on [Figma](https://www.figma.com/community/file/1428054838312216377/open-hybrid-smartwatch-app). My goal is to create a hybrid smartwatch like [this one](https://hackaday.com/2020/02/26/openchronograph-lets-you-roll-your-own-smart-watch/) which is a bit different from OSW, but I think a lot of the design/code could be shared. Let me know you think.

PS: here is the home page, with current OSW sensors (Pedometer/GPS) :
[image]

RuffaloLavoisier, 2024-10-26:
@PierreBresson 
Looks very good. Currently I'm starting to develop an app. Any ideas can give me a good direction so please find me.

fbiego, 2025-03-28:
I noticed that BLE has already been added.

I've built ChronosESP32, a library that serves as a wrapper for NimBLE to enable connectivity with the Chronos App:
https://github.com/fbiego/chronos-esp32

Features:
*  Time (Auto sync via BLE)
*  Notifications (Receive notifications from connected phone)
*  Weather (Receive weather info)
*  Controls (Music, Find Phone, Camera)
*  Phone Battery (Level, Charging state)
*  Navigation via Google Maps
*  Contacts & QR codes

NB: Chronos App is currently available for Android only.

PierreBresson, 2025-03-28:
Nice library!
@RuffaloLavoisier I've done the initial setup of the app, ble scan is working on ios/android. Here is the [repo](https://github.com/PierreBresson/watchyy).

RuffaloLavoisier, 2025-04-11:
@PierreBresson 
Nice - Thanks I will check it.