Open-Smartwatch/open-smartwatch-os #142: Feature Flags

Simonmicro, 2021-08-05:
Every part of the OS should be turned on/off during compile time, based on the hardware-**revision**.

E.g. `platformio.ini` contains the `LIGHT_EDITION_V3.3`, which includes everything except GPS and only wifi xor ble support. The `LIGHT_EDITION_V4.0` would include both and also could turn on the STA+AP mode simultaneously (as it has more flash memory).

This would allow us to support older revisions indefinitely but still add new stuff down the road. The build script must be also extended to allow us to build EVERY os-version.

The docs will include a list of enabled features for the os based on the hardware revision and when it was introduced (commit hash). It should also contain a list of all `OSW_FEATURE_*` defines available, so the user could also compile his own (unsupported) OSW-OS firmware combination.

Ideas? Add them down below.

uvwxy, 2021-08-05:
Great concept. Let me know if I can help.

mpegg4, 2021-08-11:
This really is a great idea.  would there be an overlap with config.h or would this be something config.h would override?

simonmicro, 2021-08-12:
The config.h will be able to enable further features, the platform.ini will be responsible to define the base setup.

This comes soon ™️

mpegg4, 2021-08-12:
soon ™️ my friend, soon ™️. love it thank you

simonmicro, 2021-08-19:
@uvwxy Could you try to split up the GPS_EDITION flag into those:
```
OSW_FEATURE_GPS_API -> hal
OSW_FEATURE_GPS_APP -> app
OSW_FEATURE_VIBRATOR -> api
OSW_FEATURE_SDCARD -> api
OSW_FEATURE_SENSOR_BMA400 -> api
OSW_FEATURE_SENSOR_BME280 -> api
OSW_FEATURE_SENSOR_QMC5883L -> api
OSW_FEATURE_PSRAM_GRAPHICS -> api

OSW_FEATURE_PINS_GPS
OSW_FEATURE_PINS_GPS_ROTATED
```
I have no idea how easy that is (or what you think about that) - we can also talk about that later...

simonmicro, 2022-10-03:
Merged with #190 - still needs some love by @uvwxy (GPS_EDITION), but for now done :partying_face:

closed 2022-10-03 by simonmicro