Open-Smartwatch/open-smartwatch-os #143: Platform IO Build Error

krishpranav, 2021-08-06:
Hi @Open-Smartwatch team so I am facing this error when I run this command:
```
pio run -e pico32_LIGHT_EDITION -t upload
```

I am on macOS Big Sur I have installed the requirements that you gave
```
swig, platformio
```

krishpranav, 2021-08-06:
```
src/apps/_experiments/runtime_test.cpp: In member function 'virtual void OswAppRuntimeTest::setup(OswHal*)':
src/apps/_experiments/runtime_test.cpp:14:31: error: 'MINI_IOT_DEVICENAME' was not declared in this scope
   miniIot = new MiniIotClient(MINI_IOT_DEVICENAME, MINI_IOT_SERVER, wifiSsid.c_str(), wifiPass.c_str());
                               ^
src/apps/_experiments/runtime_test.cpp:14:52: error: 'MINI_IOT_SERVER' was not declared in this scope
   miniIot = new MiniIotClient(MINI_IOT_DEVICENAME, MINI_IOT_SERVER, wifiSsid.c_str(), wifiPass.c_str());
                                                    ^
*** [.pio/build/pico32_GPS_EDITION/src/apps/_experiments/runtime_test.cpp.o] Error 1
========================== [FAILED] Took 9.47 seconds ==========================

Environment         Status    Duration
------------------  --------  ------------
pico32_GPS_EDITION  FAILED    00:00:09.470
==================== 1 failed, 0 succeeded in 00:00:09.470 ====================
```

matti04, 2021-08-06:
Hello! 
Have you removed the `.example` extension from the `configuration.h` file?

krishpranav, 2021-08-06:
oh should I do it?

krishpranav, 2021-08-06:
@matti04 I've just forgetter it i will try it

krishpranav, 2021-08-06:
compilation terminated.
*** [.pio/build/pico32_LIGHT_EDITION/lib925/Adafruit BME280 Library/Adafruit_BME280.cpp.o] Error 1
Compiling .pio/build/pico32_LIGHT_EDITION/libe10/BME280/BME280.cpp.o
.pio/libdeps/pico32_LIGHT_EDITION/BME280/src/BME280.cpp: In member function 'bool BME280::InitializeFilter()':
.pio/libdeps/pico32_LIGHT_EDITION/BME280/src/BME280.cpp:85:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
.pio/libdeps/pico32_LIGHT_EDITION/BME280/src/BME280.cpp: In member function 'bool BME280::WriteSettings()':
.pio/libdeps/pico32_LIGHT_EDITION/BME280/src/BME280.cpp:122:1: warning: no return statement in function returning non-void [-Wreturn-type]
 }
 ^
========================= [FAILED] Took 28.33 seconds =========================

Environment           Status    Duration
--------------------  --------  ------------
pico32_LIGHT_EDITION  FAILED    00:00:28.335
==================== 1 failed, 0 succeeded in 00:00:28.335 ====================

krishpranav, 2021-08-06:
In file included from .pio/libdeps/pico32_LIGHT_EDITION/Adafruit BME280 Library/Adafruit_BME280.cpp:31:0:
.pio/libdeps/pico32_LIGHT_EDITION/Adafruit BME280 Library/Adafruit_BME280.h:26:32: fatal error: Adafruit_I2CDevice.h: No such file or directory

krishpranav, 2021-08-06:
facing another error

uvwxy, 2021-08-06:
@krishpranav  please try:

`git checkout develop`
`git submodule update`

then build again.

krishpranav, 2021-08-06:
> @krishpranav please try:
> 
> `git checkout develop`
> `git submodule update`
> 
> then build again.

ok let me try

krishpranav, 2021-08-06:
Error: Please specify `upload_port` for environment or use global `--upload-port` option.
For some development platforms it can be a USB flash drive (i.e. /media/<user>/<device name>)
*** [upload] Explicit exit, status 1
================================= [FAILED] Took 115.49 seconds =================================

Environment           Status    Duration
--------------------  --------  ------------
pico32_LIGHT_EDITION  FAILED    00:01:55.487
============================ 1 failed, 0 succeeded in 00:01:55.487 ============================

krishpranav, 2021-08-06:
it is now showing this error

krishpranav, 2021-08-06:
any one know to fix it?

krishpranav, 2021-08-06:
I just run this command:
pio run -e pico32_LIGHT_EDITION -t upload --upload-port 8080

krishpranav, 2021-08-06:
File "/usr/local/Cellar/platformio/5.1.1/libexec/lib/python3.9/site-packages/serial/serialposix.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port 4444: [Errno 2] No such file or directory: '4444'

krishpranav, 2021-08-06:
I am facing this issue

krishpranav, 2021-08-06:
x.py", line 325, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 2] could not open port 8080: [Errno 2] No such file or directory: '8080'

richtepa, 2021-08-07:
> I just run this command:
pio run -e pico32_LIGHT_EDITION -t upload --upload-port 8080

Where does port 8080 come from? Is your watch is plugged in via usb?
On my mac the port is `/dev/cu.serial-*`.
Try checking the ports using `ls /dev/cu.*` in the terminal.

krishpranav, 2021-08-07:
> > I just run this command:
> > pio run -e pico32_LIGHT_EDITION -t upload --upload-port 8080
> 
> Where does port 8080 come from? Is your watch is plugged in via usb?
> On my mac the port is `/dev/cu.serial-*`.
> Try checking the ports using `ls /dev/cu.*` in the terminal.

ok thanks @richtepa

krishpranav, 2021-08-07:
I just got confused

krishpranav, 2021-08-07:
thanks for your help guys

closed 2021-08-07 by krishpranav