Open-Smartwatch/open-smartwatch-os #148: Smartwatch is unable to connect to the wifi

cptshitty, 2021-08-19:
hello, when I try to connect the smartwatch to WiFi, the whole app crashes and the watch restarts, I thought it was a hardware issue at first, but then I replaced the ESP32 and thiss issue persists... Is there any known solution to this problem?
thanks

richtepa, 2021-08-23:
On startup of wifi, the module draws to much power.
I recommend joining the p3dt discord: https://discord.gg/KADpP49G There is also a dedicated channel for Wifi-Issues.

**Software-fix:**
You could try setting `hal->setCPUClock(160);` (or 80/40/20/10) inside the `setup()` in `src/main.cpp`. This might work after the second try to connect.

**Hardware-fix:**
Paul fixed it by exchanging U10 (image below) with a LDO ME6211C33M5G (from here: https://www.aliexpress.com/item/1005001791835458.html). This should work.
[image]

simonmicro, 2021-08-23:
Issue templates with the explicit advice to check out older issues are coming...

closed 2021-08-23 by simonmicro