Open-Smartwatch/open-smartwatch-os #152: Calendar app

Kokofruits1, 2021-08-22:
[image]

I think that a calendar app could be a great addition to the smart watch. It would use the apps sync functionality (I know it is not made yet) or calDAV to get this data and then it would display them. The watch itself would only show the events name, description and location.

It would show 2 events on one page, this weeks events only,  and you would scroll through events by pressing buttons up or down. When you hold the top/bottom button it would open a  page for one of the events and show it's full name, description and location.

It has a similar design language as the #20 latest design by me, so it would have some similarities in it's design to this app.

simonmicro, 2021-09-30:
Holy sh*t! I love your design language! I've just seen this issue / idea now and i'm very impressed. I think we should invest further into making a proper UI library - but for now: Your design draft is beautiful!

Kokofruits1, 2021-09-30:
Thank you for liking my design!
For the UI library from a design perspective, I'm not a programmer, just know the basics, I would like to see standardized button label designs (compact and near the buttons), in-built icons, how selected items, loading bars and a way to input a number look.
I would like to work on a design for an alarm clock, weather app, timer and other useful utilities, if you want to work on any of these apps, I would love to help you with the design!
Also, what's happening with the android app? I heard it was being worked on, but I haven't seen anything about it in months.

simonmicro, 2021-10-04:
>  Also, what's happening with the android app? I heard it was being worked on, but I haven't seen anything about it in months.

Good question... I have no idea!

> For the UI library from a design perspective, I'm not a programmer, just know the basics, I would like to see standardized button label designs (compact and near the buttons), in-built icons, how selected items, loading bars and a way to input a number look.

Next session I plan to rework the current UI class to manage the rendering pipeline and also to allow "fullscreen progress-bars" (useful also in other areas). I think I'll add stuff like drawing some elements for that to it - but we'll see...

RuffaloLavoisier, 2022-03-23:
@Kokofruits1 
Hi, I'd like to carry out your idea. I would appreciate it if you could let me know the details.

Kokofruits1, 2022-03-26:
> @Kokofruits1 Hi, I'd like to carry out your idea. I would appreciate it if you could let me know the details.

The app would download the data from the opensmartwach app, then using the calendar permission it would download about a week worth of events to the watch. If the app is never created it could use calDAV, but I have no idea how much space it would require.

Short pressing a button would move a page up or down and long pressing a button will bring the full view for the event. When in the full view you would use the buttons to scroll or to go back. Back button could be the third button, the one on the left side, not as it is in the original design.

RuffaloLavoisier, 2022-03-26:
@Kokofruits1 

1.Does scrolling only go to the next (future) schedule ?  

2.One schedule in detail is full view (select==choice schedule)? is right?

RuffaloLavoisier, 2022-07-04:
[image]

First design 👍

RuffaloLavoisier, 2022-07-31:
Typically supported is the color, name, and time of the plan that you set in your calendar. This is limited to the Android I use, but I would appreciate it if you could check what information is displayed another mobile devices.

RuffaloLavoisier, 2022-07-31:
Also, if you want to check your week's schedule, it's common, efficient, and in practice to check it on mobile. You can check only TODAY schedule on the watch. What is currently implemented is a simple framework and a sample.

PR #279 (2022-08-06): Add OSW-APP for Calendar
Add Calendar APP
-Resolve #152
commit 5fcf505b (2022-08-06): OswAppCalendar : add calendar form

- Resolve #152
closed 2022-08-07 by RuffaloLavoisier