Open-Smartwatch/open-smartwatch-os #154: hardware

krishpranav, 2021-08-29:
hi @open-smartwatch-os team so can I know which hardware will open-smartwatch-os supports?

matti04, 2021-08-29:
Hi! 
The Open-Smartwatch OS is developed in parallel to the Open-Smartwatch. We cannot ensure and maintain compatibility with other hardware. 
The specs for the Open-Smartwatch can be found on the [respective repository](https://github.com/Open-Smartwatch/open-smartwatch-light). In short, it features an IPS display based on GC9A01, TTGO-32-micro, along with various sensors. 
While there are two major branches (light + GPS version), other makers made modifications to the current hardware, for example USB-C support or touch screen. These hardware changes might be integrated in the "official" OS at some point. 
Feel free to join our [Discord server](https://discord.gg/9DK5JY6) for further advice.

krishpranav, 2021-08-29:
thanks @matti04 for your help

closed 2021-08-29 by krishpranav