Open-Smartwatch/open-smartwatch-os #169: Battery duration

HerrAugust, 2021-11-07:
Hello, I bought the smartwatch light version (without GPS) and I will use one of the two batteries:

1. https://es.aliexpress.com/item/4000293204820.html
2. https://a.aliexpress.com/_mrvOz8P

Can anyone tell me about the estimated battery duration when the smartwatch has:

1. screen turned off
2. BLE connected all the day to an Android v8 smartphone (to received message and email notifications)
3. WiFi always turned off

Thank you a lot!

simonmicro, 2021-11-08:
We should add this to the documentation - @uvwxy didn't you created some beautiful graphs over on Discord? I'm unable to find all of them - pm me so I can help you on that 😁

@HerrAugust I think this issur more belongs to the documentation project...

simonmicro, 2021-11-08:
And I'm too stupid to move the issue to the documentation project... Help... 😳