Open-Smartwatch/open-smartwatch-os #176: Step counter

RuffaloLavoisier, 2021-11-29:
# What is not working?
A clear and concise description of what the bug is.
```
Hello, World ! 
I think device not working for ​calculate the number of steps.
Today's steps do not go to yesterday. In other words, it is initialized after a day.
```
# How to reproduce this
Steps to reproduce the behavior:
1. upload ***`OSW-OS - develop`***
2. ***`Run the step count.`***
3. ***`waiting for tomorrow.`***
4. ***`See error`***

# What should happen instead?
A clear and concise description of what you expected to happen.

# Environment
- **Model** [Light Edition 3.3]
***`Light Edition 3.3`***
- **OSW-OS** [e.g. Git commit Hash]
latest build : https://github.com/Open-Smartwatch/open-smartwatch-os/commit/04d28ca02870ff926aad8973f8f70c027c4393ea
- **Browser** [e.g. Firefox, Chrome]
***`Chrome`***

# Additional context
Add any other context about the problem here.
[image]

simonmicro, 2021-11-30:
Does the watch perform a full reset every day after midnight (and waking up from light/deep sleep)? Because for me everything works perfectly since Sunday using light sleep (maybe deep sleep is a problem - I have a feeling?)...

Could you try to catch the **full** logs before (during) and after next midnight during wakeup? So we could try to get an idea what is going on? Also I am running on 08c732f , but that should not be relevant...

_Strange behavior..._

RuffaloLavoisier, 2021-11-30:
@simonmicro I'll check it out today. Do you have a guide for the upload process? before clean or factory reset firmware(or based little test code).. I always just upload it, but I don't know if this is the wrong way.

RuffaloLavoisier, 2021-12-01:
```if (currDoW != lastDoW) {
    const uint32_t currentSteps = this->getAccelStepCount();
    this->_stepsCache[lastDoW] = currentSteps;  // write current step to last dow
    this->_stepsSum += currentSteps;            // Let's just hope this never rolls over...
    this->resetAccelerometer();

simonmicro, 2021-12-02:
You describe the correct behavior - the current step count never comes from the cache, but instead is being read from the chip. Only when the day changes we are applying the current step count to the cache and write it back as the history (and then resetting the chip). Yes, this means picking up the watch at 12 o'clock will cause all the steps before its first activation also be counted to yesterday, but this is the only solution without waking up the watch regularly (as the step counter chip is not having a step-by-day history itself). So, what is the problem (based on your opinion) here?

RuffaloLavoisier, 2021-12-02:
This is some logs after using debugging syntax in the watch face and activating the sensor properly.
``` 
src/apps/watchfaces/watchface.cpp: Watchface-log : (Today 260, Total 263) - {0, 0, 0, 263, 0, 0, 0}
```
And after a day, whether the value is **not transferred to yesterday** or **initialized becomes** as follows. 
```
src/apps/watchfaces/watchface.cpp: Watchface-log : (Today 0, Total 3) - {0, 0, 0, 0, 3, 0, 0}
```
Also, at this time, looking for where `3` comes from. A day later, `3` was recorded in the total even though the sensor was not operated.

simonmicro, 2021-12-09:
@RuffaloLavoisier could you recheck with at least commit 542922216a26e9557776711a6fac4c696a24019c ? I have added multiple sanity checks to prevent further undefined behavior...

RuffaloLavoisier, 2021-12-11:
@simonmicro As a result, it works well. Thank you very much for your hard work .
One thing need to check is that if you check the clock at `23:59:55`, the number of steps today_box will move to tomorrow_box at `00:00:00`.  However, if you press the initialization button, it will reset normally. (If you check in the master branch, the number of steps will remain as it is, and only the pointer pointing to today will be moved to initialize the number of steps today.

simonmicro, 2021-12-11:
Sure - makes sense, based on the current implementation. I guess I'll be able to fix it - we just have to execute the `setup()` code more regularly and with a cache to prevent excessive NVS interaction.

simonmicro, 2021-12-23:
Last problems are now fixed on `develop` (not the problem mentioned ↑).

closed 2021-12-23 by simonmicro