Open-Smartwatch/open-smartwatch-os #180: Display won't come on/stay on after enablng SERVICE_BLE_COMPANION

datitzer, 2021-12-02:
# What is not working?
Enabling SERVICE_BLE_COMPANION in config.h results in display issues where the display briefly comes on but goes to black. 

# How to reproduce this
Steps to reproduce the behavior:
1. uncomment and set SERVICE_BLE_COMPANION to '1' in config.h
2. build and upload firmware to open smartwatch
3. detach USB and touch reset button at least once to bring up watch
4. observe brief display before going black again

# What should happen instead?
Display should come on and remain on until timeout.

# Environment
- Light Edition
- master branch of open-smartwatch-os, with config.h changed as usual

# Additional context
none

MichaelAllenHardeman, 2021-12-06:
I have this exact same problem. I'm attempting to debug it a bit but I've never done this before. When I enable `SERVICE_BLE_COMPANION` and open the Monitor Serial IO as I flash the OS to the device I see this:

```
> Executing task: platformio device monitor <

--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Miniterm on /dev/ttyUSB0  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun  8 2016 00:22:57

rst:0x5 (DEEPSLEEP_RESET),boot:0x13 (PI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download
␁␇␀␀␀␀��␄␀␇␒ U␀␀␀␀��␄␀␇␒ U␀␀␀�Exception in thread rx:
Traceback (most recent call last):
  File "/usr/lib/python3.9/threading.py", line 973, in _bootstrap_inner

--- exit ---
    self.run()
  File "/usr/lib/python3.9/threading.py", line 910, in run
    self._target(*self._args, **self._kwargs)
  File "/home/michael/.platformio/penv/lib/python3.9/site-packages/serial/tools/miniterm.py", line 499, in reader
    data = self.serial.read(self.serial.in_waiting or 1)
  File "/home/michael/.platformio/penv/lib/python3.9/site-packages/serial/serialposix.py", line 595, in read
    raise SerialException(
serial.serialutil.SerialException: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

Terminal will be reused by tasks, press any key to close it.
```

MichaelAllenHardeman, 2021-12-06:
I tried using 3_2 env, the 3_3 env, and the 4_0 env. they all fail with an unhandled excepetion.

uvwxy, 2021-12-06:
Short Answer: BLE does not work.
Long Answer:

The BLE companion was added without testing a while ago.

What I have mentioned a couple of times on discord: the OSW does not have enough RAM to run the BLE Companion and the Display with a full buffer.
BLE leaves 80kb RAM, but we need 120kb RAM for the display data.

You are most likely running into seg faults / memory exceptions.

This is not a bug. The BLE companion is a proof of concept that is still in the code or needs to be adapted, either by reducing the display performance, or new hardware...

closed 2021-12-06 by uvwxy

simonmicro, 2021-12-06:
Hey, I would leave this open as I plan to look further into it... :pleading_face:

datitzer, 2021-12-06:
Ahh. That explains the display issue.

Thanks, Paul.

uvwxy, 2021-12-06:
> Hey, I would leave this open as I plan to look further into it... 🥺

In that case we'd need a full feature, not a bug ticket ;).

just to give you some more room to try this tech demo try calling `hal->disableDisplayBuffer();` after `hal->setup(...)`

RuffaloLavoisier, 2022-03-23:
> the OSW does not have enough RAM to run the BLE Companion and the Display with a full buffer.
> BLE leaves 80kb RAM, but we need 120kb RAM for the display data.

Is this the beginning of a solution? #202