Open-Smartwatch/open-smartwatch-os #182: Battery indicator disappear

RuffaloLavoisier, 2021-12-10:
# What is not working?
A clear and concise description of what the bug is.

# How to reproduce this
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

Follow :
insert `-D RAW_SCREEN_SERVER` in `platformio.ini` build_flag - Light Edition V3.2
build + upload 
```
Since the clock turns off after 10 seconds, 
wait for the phenomenon by inputting a button so that it does not turn off.  
Or, it disappears when you run WiFi connection. 
But it(charger indicator) does not disappear after 10 seconds when you connect the charger.
```
# What should happen instead?
A clear and concise description of what you expected to happen.

It can't disappear.

# Environment
- **Model** [e.g. Light Edition]
- **OSW-OS** [e.g. Git commit Hash] 
- **Browser** [e.g. Firefox, Chrome]
Model : `Light Edition V3.3`
branch : `develop`
Browser : `Chrome`

# Additional context
[image][image]

Is something colliding with each other? 

Add any other context about the problem here.

simonmicro, 2021-12-10:
https://github.com/Open-Smartwatch/open-smartwatch-os/blob/23232eb4ab1dd44ad7aa45463de78a7c177c2958/src/overlays/overlays.cpp#L50

This is because we can not utilitze

simonmicro, 2021-12-10:
Just comment the check and watch what starts to happen. Spoiler: The ADC does not report _any_ realistic values anymore.

RuffaloLavoisier, 2021-12-10:
@simonmicro Yes, that's right. It looked like a bug, so I thought it was a problem with the code..but It's a hardware problem.

simonmicro, 2021-12-11:
I've specifically implemented a small delay until we disable the battery indicator, so you have at least a chance of reading it before we connect to WiFi :)

closed 2021-12-11 by simonmicro