Open-Smartwatch/open-smartwatch-os #188: Reset Step Counter

RuffaloLavoisier, 2022-02-13:
I think we need to designate and initialize a specific section of the step counter. (The beginning of the week, the beginning of the month)

simonmicro, 2022-02-17:
Fair point - I have seen #189 , but I do not think this would be fine with everyone. Could you add a config option to let the user choose at least between `Monday` / `Sunday` / `No reset`? This is because, some of my watches I'm not using very often and I would like to keep my "fitness-progress" :P . Additionally here in Germany the week starts on Monday.

Thanks in advance!

RuffaloLavoisier, 2022-02-17:
> user choose at least between `Monday` / `Sunday` / `No reset`?

Good.. I see, Thank you

commit 96f3ae0c (2022-03-06): Add reset Step Counter fix #188
- add Web interface key, reset timing
closed 2022-03-06 by RuffaloLavoisier