Open-Smartwatch/open-smartwatch-os #194: Arrange the order of the main screen elements

RuffaloLavoisier, 2022-03-08:
# What is not working?
A clear and concise description of what the bug is.

[image]

* Just , we need to arrange the order.(image sort)

# How to reproduce this
Steps to reproduce the behavior:

1. Until the number of steps in blue fills up half a lap. 
2. When the minute hand is between 25 and 35 minutes.

# What should happen instead?
A clear and concise description of what you expected to happen.

* I think it would be better to see the images without each other covered. Wouldn't it be better to give priority at least?

# Environment
- **Model** [e.g. Light Edition]
- Light Edition 3.3
- **OSW-OS** [e.g. Git commit Hash]
- last version
- **Browser** [e.g. Firefox, Chrome]

# Additional context
Add any other context about the problem here.

RuffaloLavoisier, 2022-03-08:
If it is approved, I will start implementing it.

simonmicro, 2022-03-17:
Approved.

closed 2022-03-18 by RuffaloLavoisier