Open-Smartwatch/open-smartwatch-os #200: Step Counter cache problem

RuffaloLavoisier, 2022-03-20:
[image]

# What is not working?
A clear and concise description of what the bug is.

Currently, the step counter is working normally, but there is a slight problem saving the steps after the day. I'm worried about this problem.
 
As far as I know, when i infer the height of the blue square that appeared suddenly, it is confirmed that the current total steps is stored in `lastDoW`.

The `total-steps` is stored in the `lastDoW`.

we have to fix it or **wontfix**?

*I think it's a very minor problem, and I think it's a small exception and it's a part of it that doesn't cause much trouble with the operation.*

# How to reproduce this
Steps to reproduce the behavior:

1. Running and working , and check step counter

2. Please turn off the timeout and observe the step counter movement in midnight.

# What should happen instead?
A clear and concise description of what you expected to happen.

# Environment
- **Model** [e.g. Light Edition]
`Light Edition 3.3`
- **OSW-OS** [e.g. Git commit Hash]
-`last commit`
- **Browser** [e.g. Firefox, Chrome]

# Additional context
Add any other context about the problem here.

simonmicro, 2022-03-30:
Yes, this is known... Thanks for the report - will be fixed soon :tm:

RuffaloLavoisier, 2022-03-30:
This is a previous issue related to this function.
https://github.com/Open-Smartwatch/open-smartwatch-os/issues/176

PR #315 (2022-10-27): Bugfix for #200
Fixes #200
closed 2023-01-01 by simonmicro