Open-Smartwatch/open-smartwatch-os #201: Arc and Step-history color choice

RuffaloLavoisier, 2022-03-20:
[image]

When I checked after solving the problem(https://github.com/Open-Smartwatch/open-smartwatch-os/issues/194), I found that it is difficult to check if the step history color and the Arc color overlap each other (if step history is small). 

solve :  Is it better to add a separate frame so that you can check the step history?(  not bad )Or change the user's basic color to solve this problem?( I think `EASY` BETTER).

RuffaloLavoisier, 2022-03-20:
And battery color topic..

simonmicro, 2022-03-30:
We should probably finally relocate the history bars on some watch faces. This will finally solve these problems...

RuffaloLavoisier, 2022-03-30:
So i will change color..

RuffaloLavoisier, 2022-04-02:
[image]

YOLO

commit f12436ce (2022-03-30): Merge pull request #204 from RuffaloLavoisier/change_color

Watchface : fixed #201