Open-Smartwatch/open-smartwatch-os #202: Flash memory stable

RuffaloLavoisier, 2022-03-23:
```
RAM:   [==        ]  17.7% (used 57860 bytes from 327680 bytes)
Flash: [========  ]  80.2% (used 2102164 bytes from 2621440 bytes)
Building .pio/build/LIGHT_EDITION_V3_3/firmware.bin
esptool.py v3.1
Merged 1 ELF section
======================= [SUCCESS] Took 39.42 seconds ====================

Environment         Status    Duration
------------------  --------  ------------
LIGHT_EDITION_V3_3  SUCCESS   00:00:39.420
======================= 1 succeeded in 00:00:39.420 =======================
```
I solved the instability that has been raised by linking Bluetooth and Wi-Fi. 
Secure stable flash memory space.

RuffaloLavoisier, 2022-03-23:
Ping @uvwxy @simonmicro

simonmicro, 2022-03-23:
1. No need to ping, at least I get the notifications anyways.
2. What are you talking about? What did you solve? How did you solve it? What were the symptoms in the first place?

RuffaloLavoisier, 2022-03-23:
> 2\. Or do you mean that compiling both WiFi && BLE was too big for the chip?

@simonmicro 

That's right

Our chip capacity was relatively small to use Bluetooth and Wi-Fi at the same time. 

So, as far as I know, we only use one , Bluetooth or Wi-Fi. 

However, by modifying the build partitions, we have secured enough space to compile. Please make sure that what I did is worthwhile.

simonmicro, 2022-03-30:
As this would kill off our OTA capabilities, I'm closing this for now...

closed 2022-03-30 by simonmicro