Open-Smartwatch/open-smartwatch-os #206: Watchface - Dual time

RuffaloLavoisier, 2022-04-03:
Dual-time would be a good tool for people living  at different timezone. 

Instead of adding a new watchface, I think to implement additional  features to enable dual-time depending on whether you are a user or not in <Watchface analog>. Implement up to 2-3 different timezone (including the current region).

Please give me your opinion.

RuffaloLavoisier, 2022-04-10:
[image]

first attempt :laughing: 

1. Marked a common second in the middle. remove it if it is unnecessary.

2. The thing that bothers me the most is the arrangement. How can I make the space between the two sides and the top?

## Why Digital Dual Time?

The reason why we use dual time is to know different TimeZone and AM/PM and date. Analog watchface are not enough to tell you this.

## Create one watchface-Dual.

Create one more watch than add features to OSW digital watch. Adding to existing features will allow you to use either dual-time or digital watches(*only one*). Setting up dual-time whenever you need it is cumbersome. Some people who use digital watches may not need dual time. 

## PS

Please let me know if there is a better design and arrangement of ideas or elements

@simonmicro Since you are developing an emulator, i will not add it to an additional virtual graphics library repository.

mat942, 2022-04-14:
Love the idea! I would also perhaps reflect the timezone code somewhere. is there a way to store timezones in the watch (like favorites?) great reat feature, will definitely add it to my watch once released. Happy to help!