Open-Smartwatch/open-smartwatch-os #207: Data information

RuffaloLavoisier, 2022-04-04:
I would like to provide information to Watchface. 

1. Battery Notification

2. Notification of step-counter (current/total/target)

3. Distance of steps

4. Calories consumed

I think we can provide this kind of information at the current step counter.

problem related to :

* https://github.com/Open-Smartwatch/open-smartwatch-os/issues/23

RuffaloLavoisier, 2022-05-20:
See https://github.com/Open-Smartwatch/open-smartwatch-os/pull/223

RuffaloLavoisier, 2022-06-15:
It's time to restart this part.

simonmicro, 2022-08-24:
@Lorenzosciacca I think this is a GREAT idea and - from the looks - beautiful implementation! I think using openweather is OK, as e.g. https://github.com/sqfmi/Watchy/blob/5690db077f5c99c7823cb652009bc260605b231f/examples/WatchFaces/7_SEG/settings.h#L6 is also doing that. I'm very impressed!

RuffaloLavoisier, 2022-08-24:
@Lorenzosciacca If you want, could you implement this weather part?

Lorenzosciacca, 2022-08-25:
@simonmicro  thank you!
@RuffaloLavoisier , of course, it will be a pleasure for me.

RuffaloLavoisier, 2022-08-25:
@Lorenzosciacca 
Thank you in advance ☺

Lorenzosciacca, 2022-09-10:
flash update: this is how the app looks like now. While i was playing with the smartwatch the screen died, it’s the usual problem but this slowed down everything. In the next future, when I will be able to do a couple of tests, I will post everything.

https://user-images.githubusercontent.com/60114811/189479755-361d1fea-7265-46be-91ac-520c34bfa87e.mp4

mat942, 2022-09-14:
Looks great, love the UI!

Lorenzosciacca, 2022-09-27:
The app is available [here](https://github.com/Lorenzosciacca/OSW_weather).
Note: there is a bug, when i try to perform the GET to openweather i can't enable the wifi directly from the app, instead it's necessary to enable the WiFi from the webserver/time-update app. I don't know the exact reason, it also may be related to the usual WiFi problems.

RuffaloLavoisier, 2022-09-27:
@Lorenzosciacca Could you open the pull request?

Lorenzosciacca, 2022-09-27:
Yes, i can, now i'm not at home, tomorrow morning i can do It for sure. On develop right?

RuffaloLavoisier, 2022-09-27:
Yes, correct