Open-Smartwatch/open-smartwatch-os #208: Modify overlays part - Wi-Fi

RuffaloLavoisier, 2022-04-08:
# What is not working?
A clear and concise description of what the bug is.

The reception sensitivity is displayed when the smartwatch is started, but it is not actually connected. I think it needs to be modified to show only when it is actually connected.

# How to reproduce this
Steps to reproduce the behavior:

Start smartwatch and watch overlays part.

# What should happen instead?
A clear and concise description of what you expected to happen.

it is show only when it is actually connected.

# Environment
- **Model** [e.g. Light Edition]
- LE 3.3
- **OSW-OS** [e.g. Git commit Hash]
- last commit
- **Browser** [e.g. Firefox, Chrome]

# Additional context
Add any other context about the problem here.

RuffaloLavoisier, 2022-04-09:
```
Welcome to the OSW-OS! This build is based on commit d8ff9d7 from Ruffalo-develop. Compiled at Apr  2 2022 13:28:29 for platform LIGHT_EDITION_V3_3.
include/osw_config.h: Accessing key id k
include/osw_config.h: Accessing key id l
include/osw_config.h: Accessing key id s2
include/osw_config.h: Accessing key id s1
include/osw_config.h: Accessing key id s3
include/osw_config.h: Accessing key id s4
include/osw_config.h: Accessing key id m
include/osw_config.h: Accessing key id s5
include/osw_config.h: Accessing key id s6
include/osw_config.h: Accessing key id s8
include/osw_config.h: Accessing key id s7
include/osw_config.h: Accessing key id f
include/osw_config.h: Accessing key id h
include/osw_config.h: Accessing key id g
include/osw_config.h: Accessing key id r
include/osw_config.h: Accessing key id c1
include/osw_config.h: Accessing key id c8
include/osw_config.h: Accessing key id c2
include/osw_config.h: Accessing key id c9
include/osw_config.h: Accessing key id c3
include/osw_config.h: Accessing key id c4
include/osw_config.h: Accessing key id c5
include/osw_config.h: Accessing key id c6
include/osw_config.h: Accessing key id c7
include/osw_config.h: Accessing key id s
include/osw_config.h: Accessing key id o
Config loaded! Version? 2
Boot count? 5577
include/osw_config.h: Accessing key id n
src/hal/environment.cpp: Current step history (day 6, today 1281, sum 20666) is: {4590, 1402, 2869, 3759, 7923, 123, [5725]}
src/services/OswServiceTaskWiFi.cpp: [Mode] Off
src/services/OswServiceTaskWiFi.cpp: [Station] Disabled station mode...
include/osw_config.h: Accessing key id e
include/osw_config.h: Accessing key id e
include/osw_config.h: Accessing key id i
include/osw_config.h: Accessing key id a
include/osw_config.h: Accessing key id b
src/services/OswServiceTaskWiFi.cpp: [Mode] Off
include/osw_config.h: Accessing key id a
src/services/OswServiceTaskWiFi.cpp: [Client] Connecting to SSID My-Wi-Fi...
Setup Done
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
========= Memory Monitor =========
core 0 (high):	4096B of 4096B
core 1 (high):	2340B of 8192B
heap (high):	304964B of 304964B
heap (curr):	210280B of 304964B
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[... 287 lines cut ...]
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
include/osw_config.h: Accessing key id a
src/services/OswServiceTaskWiFi.cpp: [Client] Connecting to SSID My-Wi-Fi...
src/services/OswServiceTaskWiFi.cpp: [AutoAP] Inactive.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
src/services/OswServiceTaskWiFi.cpp: [AutoAP] Timeout activated: 10 seconds
[... cut ...]
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
[E][esp32-hal-adc.c:135] __analogRead(): GPIO25: ESP_ERR_TIMEOUT: ADC2 is in use by Wi-Fi.
```

RuffaloLavoisier, 2022-04-09:
I found something interesting while analyzing the logs.

When initially activated, the Auto AP will operate regardless of whether it is connected or not. However, the page will not load. After a while, AutoAP seems to be disabled and connected to Wi-Fi. However, this is also not load.

simonmicro, 2022-04-13:
This issue is caused by you enabling the screen-server, which was never properly ported into the service architecture. This is a "bug"...

closed 2022-04-13 by simonmicro