Open-Smartwatch/open-smartwatch-os #212: Progress log of features in branches

mat942, 2022-04-14:
Hi OSWers, 

I was thinking we might want to start logging whats up with master or develop given we have no releases implemented:
- what features are present in that commit?
- what is in development?

I think this could be useful for people looking at the project at the different stages in time and also for returning users (like me) after a while AFK and not keeping up on discord.

Thoughts?

RuffaloLavoisier, 2022-04-15:
Hi, there :hand: 

if you look at the last commit or git log, you can see what we did.

It is also a good idea to indicate which features have been updated on a per-version basis periodically in a `OSW docs` or `readme.md`.

simonmicro, 2022-04-15:
Okay. Let's introduce semver! We could add a `CHANGELOG.md` and call our current `master` `1.0.0`. So, when merging into `master`, the devs also have to parse the commits into a proper changelog and tag the commits... Possible - but is there any tool to enforce this? Otherwise I feel like it will be forgotten at some point...

simonmicro, 2022-10-11:
Okay, the current "policy" is that we tag our `master` on updates and create a respective release on GitHub. I know this is not a `CHANGELOG.md`, but I feel this is good enough.

@mat942 Would this satisfy your request (in that case pls close this issue), or should we try to improve this?

closed 2022-10-11 by mat942