Open-Smartwatch/open-smartwatch-os #215: Watchface-Dualtime with Analog

RuffaloLavoisier, 2022-04-15:
[image]

The minutes and seconds are the same, so only the Dual-time hours-tick is displayed. Dark color. *Question* : Then how do mark morning and afternoon(AM/PM)?

>Ignore color arc.

[image]

Both are displayed in analog. The seconds and minutes is overlaps. 
*Question* : Additionally, it is difficult to display the date and AM/PM. What if I do?

[image]

Use Watchface-MIX. Local time is displayed in analog. if you are local, it's easy to know the morning/afternoon and date without display! Dual time is displayed digital. The display clearly indicates the date, morning and afternoon.

I'm thinking, Dual-time sample in the 3 analog version. I'm thinking of the first design as a draft, but I don't think it's a bad idea. leave it up to user.

If Dual-time is enabled in analog, you want to allow the user to choose the desired shape from three designs.

Please let me know if you have any ideas

simonmicro, 2022-04-15:
> Then how do mark morning and afternoon(AM/PM)?

:shrug: 

> Additionally, it is difficult to display the date and AM/PM. What if I do?

How about a smaller text right above the clocks? This would give a neat face :smiley_cat: 

> If Dual-time is enabled in analog, you want to allow the user to choose the desired shape from three designs.

:+1:

RuffaloLavoisier, 2022-05-20:
See https://github.com/Open-Smartwatch/open-smartwatch-os/pull/223