Open-Smartwatch/open-smartwatch-os #241: Build - feature in CI

RuffaloLavoisier, 2022-06-14:
How about adding it to the CI list (`feature.yml`) to check the compilation process for features that are not supported by default?
```
Raw screen server, overlaps matrix...etc
```

simonmicro, 2022-06-15:
So you want to build every firmware in every language with every possible feature flag to determine which ones work and which do not? I do not think this is helpful, as I imagine exploding CI times!!! I think I did not understand your idea correctly.

uvwxy, 2022-06-15:
Building every combination for the develop branch would bog down the pullrequest to review/merge times

RuffaloLavoisier, 2022-06-16:
I think it would be good to add a way to check the error regarding the function flag, what do you think? I don't think it's necessary to compile all models in all languages, because the value has not already been set to in `*.ini` by default. We only want to check for abnormalities in the function.

RuffaloLavoisier, 2022-06-16:
> I imagine exploding CI times!!!

> ..would bog down the..

I think we already know how to reduce the time, Maybe IMHO so it won't increase much.

simonmicro, 2022-06-16:
Okay, something goes wrong in this discussion. Let's try an other approach: I give you the following list of example devices and flags - could you than answer me a list of builds you would be running with your idea?

**Devices**
* DevA
* DevB

**Flags**
* FlagA
* FlagB
* FlagC
* FlagD

**Platforms**
* DevA(FlagA, FlagB)
* DevB(FlagA)

So, our current CI would build:
* DevA(FlagA, FlagB)
* DevB(FlagA)
* _FlagC and FlagD never used..._

Keep in mind, that GitHub only allocates X simultanous runner for our CI, so if your device-feature-platform-build-matrix grows too big, you will still be processed sequentially.

RuffaloLavoisier, 2022-06-16:
The current CI is a `style-guide` and compile `all models` and `all languages`. I think it is my idea to suggest a way to test the currently undeclared but developed flag by adding the default model and the default language device here.
```
astyle.yml
test-OSW.yml
test-feature.yml - just only default model (ex LE 3.3V and language EN) and declare feature flag(RAW screen server .. overlap anim)..
```
Did you understand? There was an error in expressing my opinion. I'm sorry.

closed 2022-06-29 by RuffaloLavoisier