Open-Smartwatch/open-smartwatch-os #249: Time running fast

AvdHdn, 2022-06-26:
# What is not working?
I noticed the time on the watch running fast.
When synchronizing the time using wifi, the time is set correctly again.
On the debug screen I noticed 2 values for the seconds. The ESP32 value and the DS3231MA.
The ESP32 value is the one running fast.

# How to reproduce this
Steps to reproduce the behavior:
- Synchronize the time using wifi.
- Wait for any amount of time. The longer you wait, the longer the time difference will be.

# What should happen instead?
- I expect the time to not run fast.
- I expected the time to be read from the DS3231MA / be synchronized with the DS3231MA value more.

# Environment
- **Model** Light Edition [e.g. Light Edition]
- **OSW-OS** Code cloned from master branch 19-06-2022

# Additional context
This issue started after having uploaded a new version of the OSW software.
Since this was not an issue with previous software versions, I do not think this is a hardware issue.

simonmicro, 2022-06-28:
Could you checkout the branch `bug/espFleeingTime`, test that firmware and report if this helps?

AvdHdn, 2022-06-28:
I have uploaded the code onto my device.
First thing that had to happen was to synchronize the time using Wifi.
When connecting the watch to the network, a red square showed up on the left of the battery indicator.
From this point on, the updating of the screen was problematic.
Items show up from different screens etc. 
When resetting the watch, these problems disappear.

It also seems to be that synchronizing the time over Wifi does not work.
In the current build, I am not able to get it to work.

RuffaloLavoisier, 2022-06-28:
> I have uploaded the code onto my device.
> First thing that had to happen was to synchronize the time using Wifi.
> When connecting the watch to the network, a red square showed up on the left of the [...]

This is a `low-memory mode`. The screen is cleaned every 10 seconds

simonmicro, 2022-06-29:
@AvdHdn As @RuffaloLavoisier already pointed out, this little red icon is the low-memory mode, which is designed to provide a more stable experience with the (previously) crashy web interface. Turns out you need some RAM if you deal with modern browsers...

Furthermore, we did not change anything on the NTP part and for me it definitively works. I've just tested it on commit eb8231d . Could you re-verify that your watch indeed did not request an NTP update? In any case the serial logs would be seriously helpful if you suspect the issue on the OS side.

RuffaloLavoisier, 2022-06-29:
NTP does not work in AP mode.

PR #263 (2022-07-25): Bug/esp fleeing time
Fixes #249 - hopefully?
closed 2022-07-25 by RuffaloLavoisier