Open-Smartwatch/open-smartwatch-os #254: NTP update error

RuffaloLavoisier, 2022-07-05:
# What is not working?
A clear and concise description of what the bug is.

Appears when trying to update NTP. maybe 3 time in 10 time.

# How to reproduce this
Steps to reproduce the behavior:
1. Go to 'Wi-Fi connect'
2. Click on 'connect and update'

If nothing happens, repeat this process.

# What should happen instead?
A clear and concise description of what you expected to happen.

Just update local time.

# Environment
- **Model** [e.g. Light Edition]
>LE_3.3
- **OSW-OS** [e.g. Git commit Hash]
>70e1b4102c5e8d8bfbd002435421ce15868b1a0c
- **Browser** [e.g. Firefox, Chrome]

# Additional context

[image]

Add any other context about the problem here.

As NTP is updated, the update time changes to dual timezone, so internally shared functions seem to collide.

simonmicro, 2022-07-11:
Maybe a bug - PLEASE add serial logs, because I've got a feeling in my gut, that this is caused by something logged there...

RuffaloLavoisier, 2022-07-11:
```
src/services/OswServiceTaskWiFi.cpp: [NTP] Started update...
src/services/OswServiceTaskWiFi.cpp: [NTP] Update finished (time of 1657578106)!
src/services/OswServiceTaskWiFi.cpp: [NTP] Started update...
src/services/OswServiceTaskWiFi.cpp: [NTP] Update finished (time of 1657578107)!
src/services/OswServiceTaskWiFi.cpp: [NTP] Started update...
src/services/OswServiceTaskWiFi.cpp: [NTP] Update finished (time of 1657578107)!
src/services/OswServiceTaskWiFi.cpp: [NTP] Started update...
src/services/OswServiceTaskWiFi.cpp: [NTP] Update finished (time of 1657578108)!
src/services/OswServiceTaskWiFi.cpp: [NTP] Started update...
src/services/OswServiceTaskWiFi.cpp: [NTP] Update finished (time of 1657578109)!
src/services/OswServiceTaskWiFi.cpp: [NTP] Started update...
src/services/OswServiceTaskWiFi.cpp: [NTP] Update finished (time of 1657578110)!
src/services/OswServiceTaskWiFi.cpp: [NTP] Started update...
src/services/OswServiceTaskWiFi.cpp: [NTP] Update finished (time of 1657578120)!
```

The default log does not detect any anomalies...

simonmicro, 2022-07-12:
So the issue occurred at?
> src/services/OswServiceTaskWiFi.cpp: [NTP] Update finished (time of 1657578110)!      //   Action !

There should be no such problem... Maybe the function inside `esp32.cpp` https://github.com/Open-Smartwatch/open-smartwatch-os/blob/2d9384ed69ba354fe81a7487a306ae90ad75ebe1/src/devices/esp32.cpp#L54 is broken because it does not properly initialize the struct to zero before using it?

RuffaloLavoisier, 2022-07-12:
[image]
This is ERROR NTP update
[image]
This is My local-time

closed 2022-07-25 by RuffaloLavoisier